openpnp/openpnp #208: Tray feeder setup, next part should be next part that will be picked?

anthonywebb, 2016-02-02:
Here is an idea for the "next part" button in the tray feeder setup.  Currently when setting up the feeder you specify the location of the first part, then there is a "next part" button that advances the amount of the offset to the next part.

It would be handy if the "next part" button took into account the current feed count and advanced to the actual next part it was going to pick.  That way I could quickly check the indexes at the same time I checked the part locations.  When I am checking my feeder setup after picking a few boards it is tedious to have to count how many blank part slots there are to know if my next job will pull actual parts or not.  Does this make sense?

vonnieda, 2016-02-02:
Hi @anthonywebb, are you talking about adding a new "next part" button or are you referring to one of the existing buttons? If so, which one? You can identify it mousing over it and looking at the tooltip.

anthonywebb, 2016-02-02:
Hey @vonnieda you busted me trying to ask the question from memory!  OK, so I went and looked again and the whole "next part location" thing is actually on the strip feeder, not the tray feeder, so I am sure that is where you got confused.

We have had really great luck with the tray feeder, it is our go to feeder actually.  But it requires that everything is set up nice and square.  I dont know if a "next part" button on there would clutter things up, and you are welcome to tell me to get lost :) but when we go to start a new job we have all these tray feeders with feed counts in them and would like to do a sanity check before we start placing the board that the "next part" is actually occupied with a part (and that it is the part you expected)

The double benefit of this feature would be to be able to zero in on the first part, and manually override the feed count and zip to the end to ensure your tray is nice and square.

We're going to try the strip feeder once my new camera lens gets here, right now for larger strips the camera wont fit both the part and the locating holes so everything gets confused.  We'll get there :+1:

vonnieda, 2016-02-03:
Anthony,

Have you tried the Feed or Move to Pick buttons in the Feeder panel?

[image: Screen Shot 2016-02-02 at 3.59.57 PM.png]

The first will perform a feed operation and then move the nozzle to the
pick location. The second will just move the camera to the pick location
without performing a feed. I'll often hit the first and then the second to
see where the next pick is going to happen from.

Let me know if that doesn't solve it. Perhaps we can adjust what those
buttons do a bit or add an additional one.

Jason

vonnieda, 2016-02-07:
@anthonywebb Ping. Can you see my previous comment and let me know if that solves the issue?

anthonywebb, 2016-02-07:
Hey @vonnieda,  I am currently on the road and havent played with it yet, but I do think that the buttons you showed will solve much of what I was thinking.  I will pay close attention on our next run of boards in the coming weeks and see if I have any more ideas.  Thanks for the help!

closed 2016-02-07 by anthonywebb