openpnp/openpnp #210: Conveyor support

rene-dev, 2016-02-04:
For conveyor support, the driver should have a load and unload function.
after unloading, the load function waits for a new board to be fed in, and restarts the job.

vonnieda, 2016-02-04:
I have no experience with conveyors, so could you go into a little more detail as to how this would work?

* Is the load/unload responsible for moving the conveyer the right amount?
* Do we need to detect if the board load failed?
* How would you typically configure a job to do use this? Is it just a matter of setting a checkbox saying "use conveyor"?
* Do you configure how many boards to run in this fashion, or just run until the operator stops it?

Any amount of detail you can go into will help define how this feature could work.

rene-dev, 2016-02-04:
I can only speak for my machine(philips csm60) but others might be similar.
the conveyer can only go off, forward, or reverse at a fixed speed.
at the placement location is a stopper pin, which goes up.
there are 3 sensors, one at the input, one at the stopper, and one at the output.
as this is very machine specific, it could be handled in the driver, as long as there are functions for this,
which enable restarting a job.
so:
1. yes, all handled by the driver
2. it cant really fail, just needs to wait until its done.
3,4. yes, if enabled, the job repeats as long as you feed it with boards.

vonnieda, 2016-02-04:
@rene-dev This makes a lot of sense, and would be very simple to add. Thanks!

cri-s, 2016-02-04:
For left to right conveyor left hand coordinate system must be used, and the camera view for user
including mouse coordinates must be flipped horizontally.
right to left works as is, you need add machine dependend code to the jobprocessor.

vonnieda, 2016-02-04:
@cri-s Why do you need to swap coordinate systems? Wouldn't you just set the board's rotation appropriately?

cri-s, 2016-02-04:
Theoretical yes. It lowers usability because then signs are inverterted.
Board 0 must then be modified depending on board size.

rene-dev, 2016-02-05:
I dont understrand this. no matter how the conveyer/board is aligned, you set up the board offset and rotation once, and all other boards will have the same offset. does not affect any coordinate system.

cri-s, 2016-02-05:
For left to right conveyor it works as is.
I assume you know how conveyor works. The front side is fix and the back
side moves as need. Openpnp internally not uses panel center position.
Jason maybe remember on fiducials that I have insisted on that when
explaining fiducials.
As consequence of this is not possible to rotate board zero position.
Everything else works with exception of board zero.
Having board zero on center facility some things but is only half of the
solution, so changing that design decision don't really solves the problem.
There exist several workaraund to this, and all have some sort of side
effects.  The workaround with no side effect is moving the stopper manually
depending panel size. On the other side correct board placement is only
important when support pins exist, that are engaged during board clamping.
Usually this is out if possibility of lower segment machines and for clamp
pressure with at least 6 bar on several actors is needed.

I think its possible to use right to left conveyor or batch loader where
support pins are necessary and deactivate conveyor on left to right when
the up looking it sensor release
And search board begin with camera. In this case no panel support pins are
possible.

Maybe focussing at low speed and using z switch it will be corrected in sw
like on 3d printer or firepick. Who knows.
Usually support pins are needed only on bigger boards or panels having
smaller boards and as written on higher machine speed, over 3000 cph.

>
> I dont understrand this. no matter how the conveyer/board is aligned, you
set up the board offset and rotation once, and all other boards will have
the same offset. does not affect any coordinate system.

ripe909, 2021-08-20:
I see this is quite an old request...  but I will add to this from using a conveyor on a NeoDen4.

The flow would be two parts, loading and unloading.  I assume each one would have it's own pipeline, with the Conveyor configured at the machine level, and enabled/disabled on the board level.

1. Loading
- Identify where the head should first position itself to have the DOWN camera in the correct position to identify the PCB edge (probably board Y location plus board length, if using a Y oriented conveyor)
- Turn on conveyor actuator (set speed, direction, time) this will move a loaded PCB into the machine.
- Check to see if PCB edge can be recognized
- If edge is not identified, repeat conveyor actuation and retry edge detection
- If edge is identified, then the board Y location is offset by the detected edge location(assuming the PCB conveyor moves in the Y direction)

2. Unloading
- This is easier, just specify the conveyor (speed, direction, time).  This may move the PCB back out the front of the machine, which is usually used for manual loading or it may move the PCB out the back of the machine if the PCB automatically moves between machines.