openpnp/openpnp #217: Drivers should be updated to use whatever maximum resolution they can handle.

vonnieda, 2016-02-13:
Most of the drivers send doubles with two digits of precision because they were all originally based on Grbl, which could only handle two. Many of the controllers we support can handle more, so we should send with precision matching that of the controller.

At the least, Smoothie based drivers can use four digits. Others need to be investigated.

cri-s, 2016-02-14:
All are basically based on grbl and can have up to 8 digits after comma.
 It is just limited by java interface code.

markmaker, 2021-04-25:
This is covered by Advanced Motion Control, see Axis resolution:
https://github.com/openpnp/openpnp/wiki/Machine-Axes#controller-settings

For code, see also:

https://github.com/openpnp/openpnp/blob/0826ee7a826725b57cf3af680a008d0143176f87/src/main/java/org/openpnp/spi/base/AbstractCoordinateAxis.java#L117

https://github.com/openpnp/openpnp/blob/0826ee7a826725b57cf3af680a008d0143176f87/src/main/java/org/openpnp/spi/base/AbstractCoordinateAxis.java#L117-L130

closed 2021-04-25 by vonnieda