openpnp/openpnp #219: Users should be able set the order of placements

vonnieda, 2016-02-15:
Currently jobs are run in the order dictated by the planner. Users have commented previously ( https://github.com/openpnp/openpnp/issues/97#issuecomment-114353391, comments on mailing list that I can't find ) that the user should be able to influence the job processing order.

I'm trying to figure out how this could work. One suggestion from #97 is that the ordering from the table be used to plan the job. I think this makes sense, but users may need more advanced ways to change the order of placements in the job. Currently you can only sort columns - you can't apply a specific order by moving rows.

In addition, having a fixed order (instead of the planner determining it at runtime) would allow users to choose a point at which to start the job. This would be very useful for finishing jobs that failed for one reason or another.

Related to #97, we're probably moving towards just having the planner go in part height, nozzle tip order. Maybe this fixed ordering is good enough and we always just show the table in the order specified by the planner.

So, the questions are:

1. Should users be able to set / override the order of placements in a job?
2. Do we need more than just column sorting? i.e. does it need to be possible to change the position of individual rows?
3. Should the planner respect the order absolutely, even if means an unoptimized placement order?
4. If the user can set the order, should there be a way to "reset" the order to what the planner thinks is the best order? (The optimize button mentioned in #97)

vonnieda, 2016-02-15:
Idea: Include an ordinal in placement. The ordinal determines the placement order and the table order is just for visualization. Ordinal is editable by the user. Add helper buttons that sets the ordinals to the current table order and to the optimized planner order.

anthonywebb, 2016-02-15:
However it gets implemented, I feel that it should be visibly evident the list of tasks that are to be performed, and mark them complete as they are completed. This makes it easy to spot how far along the job is, what has been completed, what is coming up, and makes it easy to resume a job.

vonnieda, 2016-02-15:
@anthonywebb Indeed. That will be implemented as part of #205

mikeselectricstuff, 2016-03-04:
The only situation I can think of to need to manually set placement order is where existing large/tall parts may clash with the nozzle when placing smaller ones nearby, and  that would generally be  taken   care of with the default nozzle-then-height ordering. And even then you have potential problems if feed errors mean a part gets skipped and is done later.
Other cases would be things like  placing one of each part type to test before a run, but that should be a specific operation rather than a job re-ordering.

dandumit, 2016-03-08:
Well, I subscribe to this feature request with the option to sort by feeder for example + restart job from a certain point from the list.

On the other hand since HW design it's going to a dual head (or multi head) that would pick components in parallel I think that it's needed an optimization algorithm for parallel picking and placing.  More than that this algorithm may dictate also the order of feeders in load bank.

issus, 2016-08-18:
So as not to mess with the job planner, perhaps there could be an "is planned" flag applied to the job - if the job is started without the job planner having being run, then it will run the planner to set the order prior to placement beginning.

The "is planned" flag would be set by the user re-ordering items, or by clicking a new "plan job" button. The plan job button would run the job planner, and that would set the order value for each part.