openpnp/openpnp #221: Strip Feeder Wizard, possible bug

cri-s, 2016-02-18:
On 12mm black tape, it calc 6mm (6.35 to be precise) pitch that is wrong value.
Jogging on image, the 4mm hole is exactly 4mm distant.
Image and machine.xml is send by mail to Vonnieda .

vonnieda, 2016-02-18:
Files from Moja:

[image]
[image]
[image]

machine.xml
```
<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine">
      <park-location units="Millimeters" x="10.0" y="450.0" z="0.0" rotation="0.0"/>
      <change-location units="Millimeters" x="361.6" y="494.2" z="0.0" rotation="0.0"/>
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="22964dce-252a-453e-8106-65db104a0763" name="H1">
            <nozzles>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="69edd567-df6c-495a-9b30-2fcbf5c9742f" name="N1" pick-dwell-milliseconds="0" place-dwell-milliseconds="0" current-nozzle-tip-id="null" changer-enabled="true" changer-speed="0.0" limit-rotation="true">
                  <nozzle-tips>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d791" name="NT1" allow-incompatible-packages="false">
                        <compatible-package-ids class="java.util.HashSet"/>
                        <changer-start-location units="Millimeters" x="397.9" y="508.2" z="0.0" rotation="0.0"/>
                        <changer-begin-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="397.9" y="508.2" z="-20.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="397.9" y="526.2" z="-20.0" rotation="0.0"/>
                     </nozzle-tip>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d4d791" name="NT3" allow-incompatible-packages="false">
                        <compatible-package-ids class="java.util.HashSet">
                           <id>D2PACK</id>
                        </compatible-package-ids>
                        <changer-start-location units="Millimeters" x="481.6" y="506.2" z="-20.0" rotation="0.0"/>
                        <changer-begin-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="481.6" y="494.2" z="-20.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="481.6" y="494.2" z="0.0" rotation="0.0"/>
                     </nozzle-tip>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f42d8d791" name="NT5" allow-incompatible-packages="false">
                        <compatible-package-ids class="java.util.HashSet"/>
                        <changer-start-location units="Millimeters" x="493.9" y="516.2" z="0.0" rotation="0.0"/>
                        <changer-begin-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="493.9" y="516.2" z="-20.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="493.9" y="505.5" z="-20.0" rotation="0.0"/>
                     </nozzle-tip>
                  </nozzle-tips>
                  <head-offsets units="Millimeters" x="-21.3" y="-1.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </nozzle>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="69edd567-df6c-495a-9b30-2fcbc5c9742f" name="N2" pick-dwell-milliseconds="0" place-dwell-milliseconds="0" current-nozzle-tip-id="null" changer-enabled="true" changer-speed="100.0" limit-rotation="false">
                  <nozzle-tips>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d791" name="NT2" allow-incompatible-packages="true">
                        <compatible-package-ids class="java.util.HashSet">
                           <id>R1206</id>
                           <id>P-LCC-4</id>
                        </compatible-package-ids>
                        <changer-start-location units="Millimeters" x="514.0" y="515.5" z="-24.0" rotation="0.0"/>
                        <changer-begin-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="514.0" y="505.2" z="-24.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="513.8" y="505.2" z="-0.0" rotation="0.0"/>
                     </nozzle-tip>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d491" name="NT4" allow-incompatible-packages="true">
                        <compatible-package-ids class="java.util.HashSet">
                           <id>R1206</id>
                           <id>P-LCC-4</id>
                        </compatible-package-ids>
                        <changer-start-location units="Millimeters" x="514.0" y="515.5" z="-24.0" rotation="0.0"/>
                        <changer-begin-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="514.0" y="505.2" z="-24.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="513.8" y="505.2" z="-24.0" rotation="0.0"/>
                     </nozzle-tip>
[... cut ...]
         <soft-limit-min-1 units="Millimeters" x="-1.0E-5" y="-1.00001" z="-27.00001" rotation="-360.1"/>
         <soft-limit-min-2 units="Millimeters" x="-55.00001" y="70.00001" z="-27.00001" rotation="-360.001"/>
         <soft-limit-max-0 units="Millimeters" x="700.0" y="528.0" z="1.0E-4" rotation="365.0"/>
         <soft-limit-max-1 units="Millimeters" x="648.0" y="528.0" z="1.0E-4" rotation="365.0"/>
         <soft-limit-max-2 units="Millimeters" x="700.0" y="528.0" z="1.0E-4" rotation="365.0"/>
      </driver>
   </machine>
</openpnp-machine>

```

vonnieda, 2016-02-18:
@cri-s The images look to be pretty blurry, and there is very little contrast on the holes. Can you try putting some white tape or paper under the holes? The strip feeder uses the holes with hough circles to configure and I suspect it's not finding them in this case.

vonnieda, 2016-02-18:
Note for improvement: There should probably have some asserts on things we know are impossible or very unlikely, like a part pitch of 6.35mm.

cri-s, 2016-02-18:
I had not seen this images. On black tape contrast is good and it find all
holes.  This is not visible here.   Need see to make image itself and add
more debug info.

cri-s, 2016-02-19:
i have resolved, issue was camera calibration. I have better but uncalibrated imaes, if interested, i can add it, but after camera calibration , it works without problems.

closed 2016-02-19 by cri-s

vonnieda, 2016-02-19:
Thanks @cri-s. I have created a new issue to improve the situation: https://github.com/openpnp/openpnp/issues/224