openpnp/openpnp #223: Add a two camera view option

vonnieda, 2016-02-18:
Currently when you have more than 1 camera the camera view splits into 4 panes. Four camera systems are very rare but two camera systems are very common. It would be better if we had a two view option.

* Change view rendering so that when there are two cameras they are shown side by side or above and below each other. I think side by side is probably better since screens tend to be wider than they are taller.

cri-s, 2016-02-19:
On operative setup, 3 or 4 camera systems is commone,
1 camera for tray that have bigger field of view too, and one for tape components on other side.
Further it's not uncommon to have two cameras for tape components, one high resolution/near field of view and the other with bif field of view for greater components, and to have less travel.
Cameras are cheap and having more then one uplooking can speed up real cph.

cri-s, 2016-02-19:
Probably it should be configurable if 2 or 4 cameras should be displayed on that view, that maybe the
better solution.

vonnieda, 2016-02-19:
@cri-s To clarify: The view will still show all cameras when there are more than 2. The only difference is that if there are 2 it will show them side by side instead of showing a grid of 4 with empty spaces.

In other words; nothing will change if you have > 2 cameras. It will only look different if you have exactly 2.

cri-s, 2016-02-24:
Question, it's not better to not have multiple cameras view, and on Machine Tab where the cameras
are grouped, one additional Camera view implemented as Wizard  for machine setup ?

commit 2136878d (2016-02-29): CameraPanel now remembers the user's last selected view and sets it at startup.
Fixes #223.
closed 2016-02-29 by vonnieda