openpnp/openpnp #229: Optimization: Use cvInitUndistoryMap to precalculate the lens distortion map.

vonnieda, 2016-02-22:
From comments by @cri-s:

[19:21:26]  <cri>	you choosed to do the slow way for undistort.
[19:21:38]  <cri>	the correct way is:
[19:21:45]  <cri>	cvInitUndistortMap(intrinsic,distortion,mapx,mapy);
[19:21:50]  <cri>	on init and then
[19:22:00]  <cri>	cvRemap( t, image, mapx, mapy );

richard-sim, 2016-04-08:
GitHub has automatically merged this commit/pull request into #256. I'd have rather it had kept them separate so issues could be addressed if necessary without holding up the previous pull request, but oh well.

vonnieda, 2016-04-08:
Hey Richard, thank you *so* much for these pull requests! I am out of town
at the moment so I am behind on getting these merged, but please don't let
that hold you back. So far these look great and I will get them merged as
soon as I have some computer time.

richard-sim, 2016-04-08:
No problem! The best way to learn a codebase is usually by tackling the open issues, so happy to keep going as I explore things.

commit fcd7caab (2016-04-10): Use initUndistortRectifyMap to precompute the lens distortion map. Fixes #229. The undistortion maps are lazily computed as they are each the size of the frame and thus are too big to be saved with the calibration.
commit dea1f01d (2016-04-10): Merge branch 'develop' into feature/gcode_driver

* develop:
  Release Mats used during camera lens calibration in LensCalibration. Fixes #252
  Added a couple of missing Mat release() calls that are in #256
  Use initUndistortRectifyMap to precompute the lens distortion map. Fixes #229. The undistortion maps are lazily computed as they are each the size of the frame and thus are too big to be saved with the calibration.
  In ReferenceCamera#transformImage, adjust the size of the image before rotating so that the full rotated image will fit in the new frame. Fixes #228
commit 18fbd95c (2016-04-10): Merge branch 'develop' into feature/bottom-vision

* develop:
  Release Mats used during camera lens calibration in LensCalibration. Fixes #252
  Added a couple of missing Mat release() calls that are in #256
  Use initUndistortRectifyMap to precompute the lens distortion map. Fixes #229. The undistortion maps are lazily computed as they are each the size of the frame and thus are too big to be saved with the calibration.
  In ReferenceCamera#transformImage, adjust the size of the image before rotating so that the full rotated image will fit in the new frame. Fixes #228
closed 2016-04-10 by vonnieda