openpnp/openpnp #230: Crash in Linux related to OpenCV

vonnieda, 2016-02-23:
Discussion: https://groups.google.com/d/msgid/openpnp/45263b64-8ecc-4091-9f81-d5d87dd02679%40googlegroups.com?utm_medium=email&utm_source=footer

This is probably due to Mats not being released properly. I think we saw this before in the OpenCVDriver. Seems to be Linux specific as it's not happening elsewhere.

More info may be in #125 and #87

cri-s, 2016-02-23:
It seem releasing the Mat on new code fixes the issue.
mojalovaa1 test it's more.

vonnieda, 2016-02-23:
@cri-s Okay, thanks. Please submit a pull request if you have a fix.

cri-s, 2016-02-23:
If someone need the code in the meantime, just send e-mail.

vonnieda, 2016-02-23:
Can you just post the code here?

vonnieda, 2016-11-27:
No further reports of this, closing. Please reopen if there is still an issue.

closed 2016-11-27 by vonnieda

cri-s, 2016-11-27:
It still happen on *buntu LTS. On not LTS it works.  As LTS is the
preferred version, nu.pattern need be commented out using external linking.