openpnp/openpnp #233: libuvc compile fails

lobsteroh, 2016-02-25:
Hi, I am cloning the openpnp repository into my GitHub Desktop, select branch feature/uvc-camera, can someone see why my mvn install fails with 

`[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[40,39] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.libuvc4j.UvcLibrary
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[93,25] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[540,25] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[625,41] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera.CameraId
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[129,25] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[408,41] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[408,96] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[414,30] cannot find symbol
  symbol:   class uvc_device
  location: class org.openpnp.machine.reference.camera.LibuvcCamera
[INFO] 8 errors 
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 11.619 s
[INFO] Finished at: 2016-02-25T09:31:55-05:00
[INFO] Final Memory: 20M/223M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project openpnp-gui: Compilation failure: Compilation failure:
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[40,39] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.libuvc4j.UvcLibrary
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[93,25] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[540,25] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[625,41] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera.CameraId
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[129,25] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[408,41] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[408,96] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera
[ERROR] /Users/lobsterman/Documents/GitHub/openpnp/src/main/java/org/openpnp/machine/reference/camera/LibuvcCamera.java:[414,30] cannot find symbol
[ERROR] symbol:   class uvc_device
[ERROR] location: class org.openpnp.machine.reference.camera.LibuvcCamera`

i think LibuvcCamera.java line 40

    import org.openpnp.libuvc4j.UvcLibrary.uvc_device;

should read

    import org.openpnp.libuvc4j.uvc_device;

is this the problem?

vonnieda, 2016-02-25:
Hi Robert,

This all builds find on my machine. A few things:

* Make sure you don't have unchecked in modifications to the library.
* Make sure you've installed libuvc4j locally - it's not in Maven Central
yet.

And no, I don't think the import should change. uvc_device is declared
inside UvcLibrary, at the bottom.

You had previously submitted an erroneous pull request that had this import
change. Perhaps you have some changes that are not checked in, or should be
reverted? I'd suggest doing a clean checkout of both projects and rebuild
both.

Also, just as a note: The uvc-camera branch is not finished or released
yet, so buyer beware :)

Jason

lobsteroh, 2016-02-26:
Hi Jason, thanks, i am happy to confirm that deleting my project from my Github Desktop and checking out a fresh copy of openpnp fixed the error - as you had suggested. I just started using GitHub Desktop and I think i just managed to pile up a few doodoo changes without realizing it. all good now, getting build success with the LibuvcCamera included, and am back to including the openpnp magic into my own projects. thanks for the help and am closing the issue.

closed 2016-02-26 by lobsteroh