openpnp/openpnp #235: Nozzle Tip Runout Calibration

vonnieda, 2016-02-27:
I'm noticing a lot of runout on my nozzles and others are too. This severely degrades the performance bottom vision ( #104 ) along with general placement. We need to be able to compensate for runout by checking the runout with the up looking camera and interpolating X / Y movements.

Code for this was submitted a long time ago in #57 but it was not merged due to the complexities of the change. That code can be used as a reference, along with the good circle finding we now have throughout the system.

My initial thought on this is:
* Whenever a Nozzle is commanded to move, if it has not been calibrated it should calibrate first.
* Nozzle tip change should reset the calibration status.
* It should be possible to enable / disable calibration through the UI.
* I think most of the changes can be done by branching off of ReferenceNozzle.moveTo.

cri-s, 2016-02-27:
>Whenever a Nozzle is commanded to move, if it has not been calibrated it should calibrate first.
It should be calibrated using component, checking nozzle only have some issues.
>Nozzle tip change should reset the calibration status
This is not true, if automaitc changer is used, as in this case it is unload and load using rotation 0.
It's thinkable to use red dots/color, and filter that out in order to check if nozzle position is ok or not,
in the event nozzle was removed manually.

vonnieda, 2016-05-15:
This code is now in and working, but it's not yet documented at all. Leaving this ticket open until that's done.

vonnieda, 2016-08-30:
Note: Previous comment said this is working, but that may not be the case. More testing is needed, along with documentation and any fixes required.

vonnieda, 2016-08-30:
See comments from @cri-s on #337 for potential issues in the current implementation.

vonnieda, 2016-08-30:
Also see comments on #336 for additional issues related to angle and rotation.

vonnieda, 2016-08-31:
This feature has temporarily been removed from the configuration UI since it is not working correctly and was causing confusion. See https://github.com/openpnp/openpnp/commit/5f586166bb74a81b11a78c44a15909ebc81b2bf2

vonnieda, 2018-09-20:
@markmaker The issue was related to how OpenPnP was reading coordinates from the nozzles when the compensation was applied with regards to bottom vision. Basically, there is a stack of transforms that were being applied and it seems like some were either incorrect or backwards. The comments were not particularly clear on the error, so I think it's best that someone interested in this feature dig it out and do some testing.

ldpgh, 2018-10-13:
ReferenceNozzleTip.java (develop-branac @ line 456 ... it seems to be divison-by-0
1. develop-branch
2. enabled calibration panel in ReferenceNozzleTipConfigurationWizard.java @ line 382
3. ReferenceNozzleTip @ line 420 ... lowered 360 to 3

[image]

ldpgh, 2018-10-13:
Connect Machine
1. ReferenceNozzleTip-NT2_2MM67 is still loaded to the Nozzle-N1
MachineSetup->Heads->ReferenceHead-H1->ReferenceNozzle-N1->NozzleTips->ReferenceNozzleTip-NT2_2MM67
1. Calibrate Enabled
2. Apply button
3. Press Calibrate button 
MachineSetup->Cameras->OpenPnpCaptureCamera-OpenPnpCaptureCameraUp / Location
1. Position Camera over center of location
2. Position Tool over center of location -> NaN for X and Y ... see log below

```
2018-10-14 00:03:24 Main DEBUG: Bienvenue, Bienvenido, Willkommen, Hello, Namaskar, Welkom to OpenPnP version INTERNAL BUILD.
2018-10-14 00:03:24 Scripting TRACE: Scripting.on Startup
2018-10-14 00:03:31 ReferenceMachine DEBUG: setEnabled(true)
2018-10-14 00:03:31 GcodeDriver DEBUG: sendCommand(null, 250)...
2018-10-14 00:03:31 GcodeDriver DEBUG: sendCommand(serial://COM220 null, 250) => []
2018-10-14 00:03:31 GcodeDriver DEBUG: sendCommand(connect, 5000)...
2018-10-14 00:03:31 GcodeDriver TRACE: [serial://COM220] >> connect
2018-10-14 00:03:31 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:03:31 GcodeDriver DEBUG: sendCommand(serial://COM220 connect, 5000) => [ok]
2018-10-14 00:03:31 GcodeDriver DEBUG: sendCommand(enable, 5000)...
2018-10-14 00:03:31 GcodeDriver TRACE: [serial://COM220] >> enable
2018-10-14 00:03:35 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:03:35 GcodeDriver DEBUG: sendCommand(serial://COM220 enable, 5000) => [ok]
2018-10-14 00:03:35 ReferenceHead DEBUG: H1.home()
2018-10-14 00:03:35 GcodeDriver DEBUG: sendCommand(home, -1)...
2018-10-14 00:03:35 GcodeDriver TRACE: [serial://COM220] >> home
2018-10-14 00:03:39 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:03:39 GcodeDriver DEBUG: sendCommand(serial://COM220 home, -1) => [ok]
2018-10-14 00:03:40 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:41 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-14 00:04:41 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-14 00:04:41 GcodeDriver DEBUG: sendCommand(moveto   Z42.0000  F860, 5000)...
2018-10-14 00:04:41 GcodeDriver TRACE: [serial://COM220] >> moveto   Z42.0000  F860
2018-10-14 00:04:41 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:41 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto   Z42.0000  F860, 5000) => [ok, ok]
2018-10-14 00:04:41 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-14 00:04:41 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-14 00:04:41 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-14 00:04:41 ReferenceNozzle DEBUG: N1.moveTo((53.373000, 175.606000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:41 GcodeDriver DEBUG: sendCommand(moveto X84.2620 Y174.7980   F860, 5000)...
2018-10-14 00:04:41 GcodeDriver TRACE: [serial://COM220] >> moveto X84.2620 Y174.7980   F860
2018-10-14 00:04:44 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:44 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X84.2620 Y174.7980   F860, 5000) => [ok]
2018-10-14 00:04:44 ReferenceNozzle DEBUG: N1.moveTo((53.373000, 175.606000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:44 Scripting TRACE: Scripting.on Camera.BeforeSettle
2018-10-14 00:04:45 Scripting TRACE: Scripting.on Camera.BeforeCapture
2018-10-14 00:04:45 Scripting TRACE: Scripting.on Camera.AfterCapture
2018-10-14 00:04:45 ReferenceNozzle DEBUG: N1.moveTo((53.310000, 175.291000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:45 GcodeDriver DEBUG: sendCommand(moveto X84.1990 Y174.4830   F860, 5000)...
2018-10-14 00:04:45 GcodeDriver TRACE: [serial://COM220] >> moveto X84.1990 Y174.4830   F860
2018-10-14 00:04:45 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:45 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X84.1990 Y174.4830   F860, 5000) => [ok]
2018-10-14 00:04:45 Scripting TRACE: Scripting.on Camera.BeforeSettle
2018-10-14 00:04:45 Scripting TRACE: Scripting.on Camera.BeforeCapture
2018-10-14 00:04:45 Scripting TRACE: Scripting.on Camera.AfterCapture
2018-10-14 00:04:45 ReferenceNozzle DEBUG: N1.moveTo((53.319000, 175.300000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:45 GcodeDriver DEBUG: sendCommand(moveto X84.2080 Y174.4920   F860, 5000)...
2018-10-14 00:04:45 GcodeDriver TRACE: [serial://COM220] >> moveto X84.2080 Y174.4920   F860
2018-10-14 00:04:46 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:46 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X84.2080 Y174.4920   F860, 5000) => [ok]
2018-10-14 00:04:46 Scripting TRACE: Scripting.on Camera.BeforeSettle
2018-10-14 00:04:46 Scripting TRACE: Scripting.on Camera.BeforeCapture
2018-10-14 00:04:46 Scripting TRACE: Scripting.on Camera.AfterCapture
2018-10-14 00:04:46 ReferenceNozzle DEBUG: N1.moveTo((53.292000, 175.273000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:46 GcodeDriver DEBUG: sendCommand(moveto X84.1810 Y174.4650   F860, 5000)...
2018-10-14 00:04:46 GcodeDriver TRACE: [serial://COM220] >> moveto X84.1810 Y174.4650   F860
2018-10-14 00:04:46 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-14 00:04:46 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X84.1810 Y174.4650   F860, 5000) => [ok]
2018-10-14 00:04:46 ReferenceNozzle DEBUG: N1.moveTo((53.292000, 175.273000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:04:46 Scripting TRACE: Scripting.on Camera.BeforeSettle
2018-10-14 00:04:46 Scripting TRACE: Scripting.on Camera.BeforeCapture
[... cut ...]
2018-10-14 00:05:51 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-14 00:05:51 ReferenceNozzle DEBUG: N1.moveTo((53.373000, 175.606000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:05:51 ReferenceNozzle DEBUG: N1.moveTo((NaN, NaN, 42.000000, 0.000000 mm), 0.86) (corrected)
2018-10-14 00:05:51 ReferenceNozzle DEBUG: N1.moveTo((53.373000, 175.606000, 42.000000, 0.000000 mm), 0.86)
2018-10-14 00:05:51 ReferenceNozzle DEBUG: N1.moveTo((NaN, NaN, 42.000000, 0.000000 mm), 0.86) (corrected)
```

ldpgh, 2018-10-13:
ReferenceNozzleTip.java @ line 456 . . . overwritten by `double ratio=1.0;` to circumvent divison-by-0.

Connect Machine
1. ReferenceNozzleTip-NT2_2MM67 is still loaded to the Nozzle-N1

MachineSetup->Heads->ReferenceHead-H1->ReferenceNozzle-N1->NozzleTips->ReferenceNozzleTip NT2_2MM67
1. Calibrate Enabled
2. Apply button
3. Press Calibrate button

MachineSetup->Cameras->OpenPnpCaptureCamera-OpenPnpCaptureCameraUp / Location
1. Position Camera over center of location
2.1. Position Tool over center of location -> The original runout is back <<<<<<<<<<<<<<<<<
2.2. Expectated is to have the calibration incorporated. Maybe this assumption is not valid.

ldpgh, 2018-10-14:
It seems the basic offset between nozzle-tip and nozzle (at 0 degree / with `camera.getLocation()` as reference ) is not kept.

In ReferenceNozzleTip.java:405...415 the nozzle-tip is centered and this location is stored in `Location startLocation=location;`. The mentioned basic offset would be `location-camera.getLocation()`.

Though `startLocation` is used as reference to derive the angle-dependent offset (runout, lines 419...429), it seems the basic offset from the nozzle-tip to the nozzle is not kept outside the method `calibrate()`.

ldpgh, 2018-10-14:
To illustrate the previous comment ... some screenshots.
(1) Power Button
(2) MachineSetup->Cameras->OpenPnpCaptureCamera / CameraSpecific->Position tool over center of location ... with installed nozzle-tip has calibration enabled.

OpenPnp started ... still disconnected
[image]
Power Button pressed ... it connects the machine and issues a `home()` command

End of calibrate() methode ... right after last log-message
New postion looks nice :-)
[image]

calibrate() methode returned
Before line 243 @  ReferenceNozzle.java:moveTo()
[image]

After calling line 243 @ ReferenceNozzle.java:moveTo()
[image]

Pogramm has returned to the GUI
[image]

aneox, 2018-10-17:
I have problem with bottom light. 
When camera is using for part alignment, bottom lights turns on in Vision.PartAlignment.Before.js and turns off in Vision.PartAlignment.After.js. Works good.

When camera is using for Nozzle Tip Runout Calibration bottom lights turns on in Camera.BeforeCapture.js and turns off in Camera.AfterCapture.js. But in this case, for some reason, calibration starts before lights turns on and calibration fails. If turn on lights manually, all works good.

vonnieda, 2018-10-22:
This is a summary of the comments that were deleted from #337 and #336.

CC: @ldpgh, @markmaker 

Previous brief summary: https://github.com/openpnp/openpnp/issues/235#issuecomment-423006170

These issues have not been verified. They were posted and then due to this being a new feature I removed the feature because it sounded like there were issues. It's possible that these are not issues at all, but as I mentioned in the comment above, we should investigate and validate that the feature works well.

* There is a potential divide by zero during the final angle increment.
* There was a comment that the nozzle moves in X and Y during calibration which may introduce backlash errors. This is because we check and center multiple times to get the best centerpoint. This is required when the camera is not calibrated, but maybe it would be better if we avoid it. Worth testing both ways.
* Concern that the DROs are incorrect after calibration, or that the "capture coordinates" buttons do not get valid coordinates. Should be easy to check by introducing a large calibration offset.
* A comment, "Nozzle position center is not used and as result calibration has offsets.". I'm not sure what this meant and never got an explanation.

So, given the concerns above, what really needs to happen is someone take ownership of this issue / feature, finish developing it and really test it. Don't consider the current code to be ideal - perhaps it can be much better. Maybe the algorithm can be improved. Consider the current code to be a starting point. My initial implementation of it was just a first attempt, so maybe we can do much better.

netzmark, 2018-12-15:
I have unlocked the calibration panel in the ReferenceNozzleTipConfigurationWizard to play with the feature a bit and consider why can't see the fields X,Y,Z,Rotation in the Calibration section?
Was this added by ldpgh or markmaker while their testings?

netzmark, 2018-12-15:
I can't see x,y,z,c fields (windows) not values inside them.
Should I see an explanation around the comment above?

ldpgh, 2018-12-15:
Are you running #openpnp IRC?

ldpgh, 2018-12-15:
The TODO comment in `ReferenceNozzleTipConfigurationWizard.jave` is not comming from me.

netzmark, 2018-12-15:
I didn't use irc of openpnp yet but we can try after the weekend.
On mobile now.

ldpgh, 2019-01-02:
I would like to prevent to [take over the ownership](https://github.com/openpnp/openpnp/issues/235#issuecomment-431827782)  of this issue, but I can prepare a PR.

Changes will be ...

1. Rotation parameter for calibration
1.1. Convert `angleIncrement` hard-coded into XML-parameter.
1.2. Add  parameters  `angleStart` & `angleStop `as companion to `angleIncrement` as described in the [feature request #784](https://github.com/openpnp/openpnp/issues/784).
1.3. Make those parameters accessible by machine.xml. I need those additional parameters to overwrite the SrcCode settings. My local implementation is limited to an rotation of +/-100deg. 
1.4. Add TODO comment to add angle-parameter fields to the calibration panel.

2. Subtract the Camera-Location from the new location as reported [above](https://github.com/openpnp/openpnp/issues/235#issuecomment-429617753)

3. Workaround of the Division-by-zero as reported [above](https://github.com/openpnp/openpnp/issues/235#issuecomment-429578368). It is not a fix. and will just skip the error condition. Further work is necessary.

4. Keep the pretty verbose debug-messages ... at least at the beginning.

5. Return nozzle/nozzle-tip after calibration to start-position (usually 0 degree).

mgrl, 2019-01-13:
Made an attempt to fix this in https://github.com/mgrl/openpnp/tree/runoutFix 

The principle is to fit a circle into the runout path by means of kasa algorithm. Should work in theory, will test on the machine in the next hours... It still needs cleanup, optimization and of course verification. Maybe others want to test, too. Or somebody finds obvious mistakes?

netzmark, 2019-01-13:
I'll be happy to join to the tests since tomorrow.
But could you pls to place here one final debug picture with found and drawn the contour, and place the pipeline you use? Do you search for an outer contour or the inner hole?

mgrl, 2019-01-13:
currently testing the pipeline - using the built-in pipeline for now.

netzmark, 2019-01-13:
Ok. But still actual question "outer contour or the inner hole of the nozzle"?

ldpgh, 2019-01-13:
> Ok. But still actual question "outer contour or the inner hole of the nozzle"?
As far as I can tell ... both are working. Just circle center point is used.

netzmark, 2019-01-13:
Sure they both must be working. I was just curious what others use and find as more effective...

ldpgh, 2019-01-13:
How about IRC right now ?

mgrl, 2019-01-13:
so here a capture of the measured points plus the results of the circle fitting by means of the kasa method: 
[runout export.zip](https://github.com/openpnp/openpnp/files/2753308/runout.export.zip) (libreoffice calc)

it seems the circle is well approximated, but if I apply the correction (see updated repo https://github.com/mgrl/openpnp/tree/runoutFix) there will still be offsets - I have too investigate where these may come from...

ldpgh, 2019-01-13:
MyTestProcedure-1:
- Run the Calibration
- Go to BottomCamera and jog C for (angleStart->angleStop by angleIncrement)

Do we have more simple tests?

mgrl, 2019-01-14:
so I think I got the math right, but it's not very stable yet... have to find edge cases...

but in general I think I'm on the right way...

no compensation:
[image]

with compensation:
[image]

netzmark, 2019-01-14:
Wow, congrats, it looks really good :-).
btw, your non compensated movie presents the runout like the best nozzle that I have :-) :-), all the rest are much much worse...
How looks the no-stability problem you say?

netzmark, 2019-01-14:
So it is not problem of your code but global.
I will be trying with hought circles for fiducials a bit later today.
I think it may be also good to check it for the nozzles, both cases we need to detect good circles. Unfortunately, for nozzles, an outer diameter is changing a lot between 502 nad 506-507. That is what I've asked you what do you detect, outer or inner edge, because inner diameter changes less than outer.

mgrl, 2019-01-14:
for a 505 I just used this pipeline, that seems to work stable for my machine: 
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="9" enabled="true" settle-first="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Bgr2Gray"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.BlurMedian" name="4" enabled="true" kernel-size="5"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="3" enabled="true" diameter="250"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="2" enabled="false" threshold="15" auto="false" invert="false"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="5" enabled="false" diameter="100"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="resultsb" enabled="false" threshold-step="5.0" threshold-min="0.0" threshold-max="60.0" repeatability="2" dist-between-blobs="5.0" color="false" color-value="0.0" area="true" area-min="20.0" area-max="5000.0" circularity="false" circularity-min="0.800000011920929" circularity-max="-1.0" inertia="true" inertia-ratio-min="0.5" inertia-ratio-max="1.0" convexity="true" convexity-min="0.4" convexity-max="1.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DetectCirclesHough" name="results" enabled="true" min-distance="50" min-diameter="100" max-diameter="200" dp="1.0" param-1="150.0" param-2="20.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="0" enabled="true" image-stage-name="9"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="7" enabled="false" key-points-stage-name="results">
         <color r="255" g="0" b="204" a="255"/>
      </cv-stage>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawCircles" name="6" enabled="true" circles-stage-name="results" thickness="1"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="8" enabled="true" prefix="runoutCalibration" suffix=".png"/>
   </stages>
</cv-pipeline>

and the result: :100: 
with compensation:
[image]

without compensation:
[image]

netzmark, 2019-01-14:
Ech, forgotten we can use specific pipeline to specific nozzle tip...
So now you have used hough circles and seems be stable.
Before not stable was with default SimpleBlobDetector?

mgrl, 2019-01-14:
I have used simpleBlob before for the small nozzles, but sometimes it didn't find anything. The houghCircle seems to work now properly if tuned for the size of the circle it should detect. Otherwise the circles are jumping and this gives bad results...

I submitted a pull request: #804. Maybe you could test if it works for you in general and then we will see whether we can go on or just disable the feature again until it's really really fixed...

netzmark, 2019-01-14:
Sure, I'm just going to apply your changes to my machine code and try to test it tomorrow.

netzmark, 2019-01-14:
Michael, you must have seen this for sure...
https://github.com/openpnp/openpnp/pull/803

I'm going to test both approaches tomorrow and give some feedbacks.

mgrl, 2019-01-14:
I dont really know what the differences are in #803 since the files seem to be exchanged in total. I guess both approaches won't work together

ldpgh, 2019-01-14:
It seems space/TAB respective line-ending has been changed by Eclipse.
Turn-off space, TAB, line-endings for the diff ... it should show what you like to see.

netzmark, 2019-01-14:
No need to have them working together, but test them both to find which version works better. I just have compiled two separated jars, one using your approach and second using @ldph code. Tomorrow I'm going to try to play with them both on my machine.
Another story is Jason's opinion what he finds as better :-).

vonnieda, 2019-01-14:
@mgrl @ldpgh @netzmark As you all have been working on this I'd really appreciate if you can all come to an agreement and produce one complete PR, or at least if it has to be multiple PRs make sure they don't conflict with each other. I'm not in a position right now to really test any of this, so I need to take someone's word for it that it works :)

netzmark, 2019-01-14:
To be honest, It's not my job at all, I can only join to the tests. 
These are quite differential codes, I think not possible or sensual to connect them but need to find which works better (and it also will be not very easy) and merge it to the official release. What shall we do if both work well? 😀

Michael, @ldpgh, maybe it's worth to you both do the same that me and try to test both versions...

mgrl, 2019-01-14:
I just skimmed through #803 (thanks for advice hiding whitespace diffs). It seems @ldpgh added some parameters, etc, but did not change the underlying math? Actually I didn't dive too much into the code/algorithm how it was until today, since it was blamed not to work. so I implemented my own version with the math of which I think how the mechanics work. the animated images proof that the math seems to do it's magic right ;)

I will try the changes from @ldpgh tomorrow, too.

anyway. further tests are needed of course. thats why I send the pull request...

ldpgh, 2019-01-14:
Can't say it better ... "anyway. further tests are needed of course. thats why I send the pull request..."

@vonnieda ... agree with you.

ldpgh, 2019-01-14:
Have fixed the issues of my PR (line-endings for diff, variables renamed to pass Travis-CI).
Now the diff-tool can show you the changes easily.
Test on the machine still pending ... too late today.

mgrl, 2019-01-15:
## without compensation
[image]

## with ldpgh's compensation
[image]

## with mgrl's compensation
[image]

for reference here is the used pipeline and the debug image-output to see that the pipeline detected in both cases the right center of the nozzle.
[calibration-stage.txt](https://github.com/openpnp/openpnp/files/2759557/calibration-stage.txt)
[runout-ldpgh-rawimages.zip](https://github.com/openpnp/openpnp/files/2759558/runout-ldpgh-rawimages.zip)
[runout-mgrl-rawimages.zip](https://github.com/openpnp/openpnp/files/2759559/runout-mgrl-rawimages.zip)

how did it work for you? I found still a little runout in the compensation method of ldpgh (but better than without).

markmaker, 2019-01-15:
Hi

@mgrl  / @ldpgh very unfortunate you both invested precious work into this.

Just from understanding the calculation method and commented code better, I slightly favour #804. I hope you understand, @ldpgh. 

I feel this solution is more expandable in the future, as all the fitting is done **after** the complete vision phase in a separate math phase here
https://github.com/openpnp/openpnp/pull/804/files#diff-3903f2851bc6f48dda1b8e8d72c06b40R461
and here
https://github.com/openpnp/openpnp/pull/804/files#diff-3903f2851bc6f48dda1b8e8d72c06b40R511
E.g. if you decide to add ovality later, it can be done without changing/breaking much. 

One suggestion would be to add the number of vision measurements to the UI Panel and then control `angleIncrement` with it. 

I also suggest defaulting this to a prime number (e.g. 7) plus a non-zero start angle, so no aliasing with the video pixels happens. Most cameras have some image sharpening built in, which effectively shifts any edges to the next pixel boundary, which may matter with these small offsets.  So you need to even that out by using angles that do not add up to algebraic angles. 
https://en.wikipedia.org/wiki/Trigonometric_functions#Algebraic_values 

_Mark

mgrl, 2019-01-15:
I don't see any problem to integrate both versions into one branch one day.
acutally it should not need any other compensation model, since the runout per mechanic must be a circle, must not?

I made some placements on the demoboard, no bottom camera vision:
## without compensation
[image]

## with mgrl compensation
[image]

I guess I need a nozzle with much runout :laughing:

netzmark, 2019-01-15:
I've tried to perform some tests today but had no too much time for it. However I made some and firstly found that I have serious problem with the light volume to properly detect the nozzle tip. I had to increase exposure seriously to get any result but still get many index0 (no find) answers. Or if I set houghCircles wider I often find the circles with other size I need an moved from the centre - so my process for nothing.
I will try more tomorrow. However I think that it could be usable to finally add some scripts calling or actuators before/after calibration to can control the power of lightning, or auto increasing /decreasing the exposure around the calibration.

markmaker, 2019-01-15:
> acutally  it should not need any other compensation model, since the runout per mechanic must be a circle, must not?

Within the precision that matters (and/or what vision resolution can get) you're probably right. Theoretically the axis could be tilted, resulting in a slight ovality. But then again vision gets the nozzle tip in one plane but placement on the PCB will be in another plane (due to spring loaded tip) and the ovality would actually work the other way. 

_m

mgrl, 2019-01-15:
so now I tested to go lower with the number of measurements. it seems the numbers are quite stable up to angleIncrement = 120°. especially the radius is important as it represents the runout. 3 captures are required to fit the circle (which is true for 120°). I can go this low with my implementation, since I fit a circle into the measurements and thus can calculate the exact offset for any possible angle later. The math before did just linear interpolation between two measurements and will be more unaccurate if the the number of measurements is lowered.
additional averaging for my method is not needed since the kasa fit algorithm does its best to fit the circle.

##angle increment 5° (72measurements)
[runoutFix]calculated nozzleEccentricity: centerX: 0.1696830100348604 centerY: 0.22550912581345472 radius: 0.14118548648603493
##angle increment 15° (24measurements)
[runoutFix]calculated nozzleEccentricity: centerX: 0.18089478065314937 centerY: 0.17026865964700014 radius: 0.1401856592249164
##angle increment 60° (6 measurements)
[runoutFix]calculated nozzleEccentricity: centerX: 0.18855007699737564 centerY: 0.1735530454622073 radius: 0.13638914441623307
##angle increment 90° (4 measurements)
[runoutFix]calculated nozzleEccentricity: centerX: 0.16668617902522648 centerY: 0.20960057904441326 radius: 0.13382306133470973
##angle increment 120° (3 measurements)
[runoutFix]calculated nozzleEccentricity: centerX: 0.17139662707281267 centerY: 0.222508790619469 radius: 0.14785716127571277
[runoutFix]calculated nozzleEccentricity: centerX: 0.17649940371794873 centerY: 0.22075060000000002 radius: 0.14390923565418187

ldpgh, 2019-01-15:
Did a [calibration movie](https://github.com/ldpgh/rasenmaeher/blob/master/openpnp_btmVisionCal.mp4) ...

mgrl, 2019-01-15:
Looks good. I went through the old code and it seems, the only thing that was wrong is that it centered the nozzle before calibration... That's fixed with your code. Anyway, how to go on? Shall we implement both approaches? We need to define the functions then and just switch between them...

ldpgh, 2019-01-15:
Our 2 solutions
- table-based + linear interpolation at run-time ... maximum flexibilty, can be tailored at run-time to some degree (eg. lower step size during calibration to reduce interpolation issues)
- model-based ... with model hard-coded in the sources, but less sensitive to measurement errors during calibration. In addition it will not have an overflow issue, if the rotation exceeds the calibration range.

---

With both lists above I really vote to keep both solutions in parallel. Based on its nature the solution are able to address different difficulties.

It is just one more field in  `ReferenceNozzleTipConfigurationWizard.java` to select one or the other approach.

---

When it comes to implementation ...
- `ReferenceNozzle.java`
   - In order to switch between the 2 bottom vision calibration implementations I think those modifications should be in `ReferenceNozzleTip.java`
   - ... if possible
- `ReferenceNozzleTipConfigurationWizard.java`
   - keep my changes. It has your modifications. 
   - I really like to keep the 0degree-offset shown in panel to easily qualify my different nozzle tips.
   - Implemenation relies just on:
      - isCalibrated()
      -  nozzleTip.getCalibration().getCalibratedOffset(rot);
- `ReferenceNozzleTip.java` 
   - @ ldpgh
      - There is just one method added to allow averaging around `findCircle()`.
      - All other stuff should be just "fixes".
      - adding some parameters (`angleStart`, `angleStop`, `angleAverage`) as companions to `angleIncrement`
   - @ mgrl
      -  ... if possible add some notes here in my comment ...
      - Btw.  `angleIncrement` can be changed in machine.xml already.

There is big question mark to me how to handle 2 separate implemenations in parallel for a single API in Java. I'm more a spare time java programmer.

mgrl, 2019-01-16:
Hi, okay, further we need an documentation in the wiki that gives the users the ability to make a reasonable choice about which algorithm to use and what the parameters/calculated numbers mean. what are limitations, when works which method better... (maybe a chapter here https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Nozzle-Setup or a separate site).

> I really like to keep the 0degree-offset shown in panel to easily qualify my different nozzle tips.

maybe we can rename the field to "runout" or "radius" and you calc the r=sqrt(x2+y2). that field I could use, too.

how I would implement it for now is:
 - remove other result processing than houghcircles from findCircles, but make the processing of houghcircles good, catch errors, etc. in my testing houghcircles worked well, the other method not that good. if there is a need, these functions can be readded later.
- expose angleIncrement to the user in form of number of measurements that shall be taken
- angleStart / angleStop I would leave out, as it seems this is unique to your machine. anyway it remains to be changeable in the machine.xml.
- I would try to isolate our methods into functions that can simply be switched by an if. which method to be taken is dependent on what the user chose.

If you allow, I would try to integrate your changes into my branch and do some cleanup? I don't really know how to do else...

ldpgh, 2019-01-17:
> I tried to integrate both algorithms but the way I do it with my limited java capabilities makes it quite messy.

If your algoritm is not covering this limitation ... no problem. In this case the user chooses the original table-based implementation.

> Why has your machine only +-90° nozzles? That's the only case, that my algorithm can't handle since the kasa fit works better for full circles...

It is an resolution limit.
With `pick @ -90deg and place @ +90deg` and `pick @ +90deg and place @ -90deg` the whole 360deg are covered. In reality the limit is +/-100deg to have margins for misaligned parts in the SMD-strip.

mgrl, 2019-01-17:
okay, I'm on the way... it's getting complexer and complexer 😞  

I got both approaches integrated into one now. 

First makeup for the GUI is done, too:

[image]
(switching doesn't work yet, but under the hood both algorithms coexist and should be usable).  will continue soon, but have to come to an end

mgrl, 2019-01-18:
ignore me, these are for wiki documentation
[image]
[image]
[image]

mgrl, 2019-01-18:
and here the current gui:

[image]

mgrl, 2019-01-18:
[this is for the wiki](https://github.com/openpnp/openpnp/wiki/Runout-Compensation-Setup)
[image]

[27 more replies]

commit a96efa2e (2019-01-27): first attempt to fix #235
closed 2019-01-27 by vonnieda