openpnp/openpnp #241: Jog controls: Z-Up-Button jogs Z-Axis down

Pixopax, 2016-03-03:
The Z-Up-Button jogs the Z-Axis down, Z-Down jogs it up.
Should be reversed.
Tested it with a tinyG as driver.

vonnieda, 2016-03-03:
@Pixopax This is a misconfiguration of your TinyG / machine. You just need to invert the axis in the TinyG config. Make sure that sending Gcode to move negative goes down and positive goes up.

Pixopax, 2016-03-03:
But the homing cycle reacts right...How can that be?

vonnieda, 2016-03-03:
That will need to be inverted as well. Perhaps you are using the min endstop when you should be using the max? If you post your TinyG config I may be able to help more.

Pixopax, 2016-03-03:
If I invert that the homing cycle starts with driving my Z down until it hits the table, as expected.

vonnieda, 2016-03-03:
TinyG homes towards the enabled switch. See:
https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Setup-and-Troubleshooting#axis-starts-and-stops-and-moves-in-the-opposite-direction-before-searching
and
https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Description-and-Operation

I'm not sure what machine you are working on, so I can't give detailed instructions, but the basic idea is that if you have a home switch is that at a positive (up) position you should enable the "max" switch. If you have a home switch at a negative (down) position you should enable the "min" switch. Only one should be enabled. TinyG will home towards the enabled switch.

So, set your motor inversion so that negative moves down and then once that's sorted set your switch configuration to either min or max so that the head homes in the right direction.

If you'd like more details please post your TinyG configuration and some info about your machine and switch placement. A picture would probably do the trick.

Pixopax, 2016-03-04:
My machine is a liteplacer, the switches are NO-Switches (I Know, that is not good, but the inventor did it so). 

So when I use the liteplacer setting, Openpnp drives my Z-Axis up on homing, until it hits the switch, then it does Y- and X-Axis. Allright so far. 
But as soon as I want to jog the Z-Axis down, I need to push the UP-Button instead of DOWN. 

The switches are wired as you said,

If I reverse the Z-Axis, homing drives my Z-Axis down until the min-Switch triggers. Then the UP-Jogging works in Up-Direction, but Homing is wrong.

vonnieda, 2016-03-04:
@Pixopax Sounds like it's pretty close. When you say "If I reverse the Z-Axis, homing drives my Z-Axis down until the min-Switch triggers. Then the UP-Jogging works in Up-Direction, but Homing is wrong." what do you mean that homing is wrong? Is it leaving the nozzle in the lowered position?

Pixopax, 2016-03-04:
Well, Homing Z should normally drive the Z-Axis up until it hits the limit switch right?
If I do it now, homing does what it is supposed to do: Drive the axis up until it finds the switch.
In this case, jogging the Z-Axis up by hitting the UP-Button in the Jog-Window drives my Z-.Axis down, wrong way :-(

If I reverse the axis in tinyG, and then try to home it, the homing cylcle is reversed: It drives the Z-Axis down, instead of up. But then the Joging direction is right, Up-Button drives the Z-Axis up. I cannot have both as it seems. Either homing or jogging has the wrong direction.

So somewhere, there is an error. The liteplacer software runs fine here, Up-Jogging drives the axis up, homing Z drives Z up too.

vonnieda, 2016-03-04:
@Pixopax Yes, you need to change the switch you home to along with inverting the motor direction.

See this document: https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Setup-and-Troubleshooting#homing-setup-and-configuration

It looks like you need to send $xsn or $xsm and change which switch you are using for homing. Just look at which one is currently set and change it to the other.

closed 2016-03-22 by vonnieda