openpnp/openpnp #244: Move head to a certain position + signaling when job complete or error

dandumit, 2016-03-08:
Hi,
Would be nice to have option to move placement head to a certain position when job complete and also to signal machine status acustic and optic : 
[http://www.tme.eu/gb/details/ks-ad3dczp125ryg/signalling-columns/w2/ks-ad3d_cz_p_125rygm/](http://www.tme.eu/gb/details/ks-ad3dczp125ryg/signalling-columns/w2/ks-ad3d_cz_p_125rygm/)

Br,
Daniel

vonnieda, 2016-03-08:
The first part of this is a duplicate of #76. I've modified the issue to only include the unique part.

vonnieda, 2016-04-23:
This should probably be done with a boolean actuator. Let the user set an actuator name to trigger at job end. This makes it easy to add without driver modifications.

pfried, 2016-08-30:
I actually took a look at the code and wondered if i could contribute somewhere as this seems to be a reasonable starting point.

I would possibly have introduced a Signaler Class (and an implementation as a SoundSignaler or a driver output), because the Actuator interface implements the HeadMountable Interface and also seems no good fit to me. This would provide the possiblity to signal things like:

- Feeder close to empty
- Errors (like machine stop, etc.)
- Machine running and placing

But the major blocker i see here is that there is no central "error management". If an error occurs there is no way right now to detect the problem. The only thing that happens is a log entry which is helpful for  human, but no code-wise (Maybe it would make sense to introduce error codes or something similar).

I think it would be reasonable to extend the MachineListener Interface and introduce something similar to the `fireMachineHeadActivity` but for errors or at least a `machineError` function. Of course there are many places where one would need to insert it.

Any ideas?

vonnieda, 2016-08-30:
During a job run, all errors bubble up to the JobProcessor and are eventually passed on to the UI from there. The JobProcessor would be a good central place to handle this.

I'd be okay with a Signaler class so that we could have something like a SoundSignaler, but for anything related to communicating with the outside world, it should just hand it off to an Actuator. The Actuator interface is designed for handling this kind of stuff, and I don't think this feature is worth a Driver modification which will break all existing drivers.

So, there could be a Signaler interface with a variety of potential events, and there could be SoundSignaler where you configure different sounds to play for the events, or ActuatorSignaler where you configure Actuator names to handle the events.

pfried, 2016-08-31:
Ok i see, so i would catch the exception within the ReferenceJobProcessor where it calls next on the fsm and rethrow the exception?

But this will tie the handling to this specific JobProcessor, so i would need to do the same for the PasteDispenser and the GlueDispenser processors?

vonnieda, 2016-08-31:
Yep, it would need to be done for all three. I think it would make sense to
put the code in the common base class AbstractJobProcessor so it doesn't
have to be duplicated, and then could just be called from each.

I'm open to other ideas too.

vonnieda, 2016-11-06:
This was fixed with @pfried's additions of the Signalers.

closed 2016-11-06 by vonnieda