openpnp/openpnp #246: GUI disappears after few minutes ... executable still running

ldpgh, 2016-03-25:
Have a fresh install (very 1st install) of
  OpenPnP-windows-develop.exe  (dev, Windows 32Bit, downloaded  25Mar2016)
  WinXP Home Edition w/ Sevice Pack 3

GUI disappeared after few minutes while  OpenPnP.exe  is still running in the TaskManager (about 250MByte footprint).
- Happened during 1st execution while setting up the camera (openCV camera)
- again during 2nd execution ... camera setup done ... while turning on the camera in the window. Have seen some pictures for few seconds ... then GUI disappeared again.

Is there any logFile to check?

vonnieda, 2016-03-25:
Hi @ldpgh,

Check here for the location of log files: https://github.com/openpnp/openpnp/wiki/FAQ#where-are-configuration-and-log-files-located

A few things to do:
1. Check if the log file has an error or exception near the bottom and please paste that in.
2. Paste in your machine.xml. You can find this using the same link above.
3. Please report how much memory your laptop has.
4. Try saving or backing up your config and then deleting the .openpnp directory. Reopen OpenPnP. You should see the fake test camera and the UI. Does it still crash with this config? If not, that will help narrow this down to an issue with the OpenCvCamera.

Thanks,
Jason

ldpgh, 2016-03-25:
Laptop runs with 2GByte RAM.

File "C:\Dokumente und Einstellungen\<user>\.openpnp\machine.xml"  as  zip-archive to comply with github :-( ...  [machine.zip](https://github.com/openpnp/openpnp/files/189879/machine.zip). 
According to the timestamp it seems the file was written after the camera setup has been finished successfully.

There are 3 more error-logs in the installation directory "C:\Programme\openpnp"

- "hs_err_pid2072.log" ... timestamp matches the second start of the program (aborted using task manager)
> . . . . .
> Stack: [0x133f0000,0x13440000],  sp=0x1343fa78,  free space=318k
> Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
> C  [opencv_java2411.dll+0x6e5c9]
> C  [opencv_java2411.dll+0x1475]
> J 1522  org.opencv.core.Mat.n_delete(J)V (0 bytes) @ 0x00e0d69b [0x00e0d610+0x8b]
> J 1521 C1 org.opencv.core.Mat.finalize()V (12 bytes) @ 0x00e0d431 [0x00e0d400+0x31]
> J 2208 C1 java.lang.ref.Finalizer.runFinalizer(Lsun/misc/JavaLangAccess;)V (62 bytes) @ 0x00dc7a24 [0x00dc78a0+0x184]
> J 2207 C1 java.lang.ref.Finalizer.access$100(Ljava/lang/ref/Finalizer;Lsun/misc/JavaLangAccess;)V (6 bytes) @ 0x00dc7e52 [0x00dc7e40+0x12]
> j  java.lang.ref.Finalizer$FinalizerThread.run()V+45
> v  ~StubRoutines::call_stub
> V  [jvm.dll+0x158ba5]
> V  [jvm.dll+0x21e12e]
> V  [jvm.dll+0x158c3e]
> V  [jvm.dll+0x158dc6]
> V  [jvm.dll+0x158e37]
> V  [jvm.dll+0xfd84f]
> V  [jvm.dll+0x17c07c]
> V  [jvm.dll+0x17c16a]
> V  [jvm.dll+0x1c06f6]
> C  [MSVCR100.dll+0x5c556]
> C  [MSVCR100.dll+0x5c600]
> C  [kernel32.dll+0xb713]
> C  0x00000000
> 
> Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
> J 1522  org.opencv.core.Mat.n_delete(J)V (0 bytes) @ 0x00e0d657 [0x00e0d610+0x47]
> J 1521 C1 org.opencv.core.Mat.finalize()V (12 bytes) @ 0x00e0d431 [0x00e0d400+0x31]
> J 2208 C1 java.lang.ref.Finalizer.runFinalizer(Lsun/misc/JavaLangAccess;)V (62 bytes) @ 0x00dc7a24 [0x00dc78a0+0x184]
> J 2207 C1 java.lang.ref.Finalizer.access$100(Ljava/lang/ref/Finalizer;Lsun/misc/JavaLangAccess;)V (6 bytes) @ 0x00dc7e52 [0x00dc7e40+0x12]
> j  java.lang.ref.Finalizer$FinalizerThread.run()V+45
> v  ~StubRoutines::call_stub
> . . . . .
> 

- "hs_err_pid3212.log" ... timestamp matches 2nd start of the program (aborted using task manager)
> . . . . .
> Stack: [0x133f0000,0x13440000],  sp=0x1343f888,  free space=318k
> Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
> C  [opencv_java2411.dll+0x6e5c9]
> C  [opencv_java2411.dll+0x1475]
> J 1629  org.opencv.core.Mat.n_delete(J)V (0 bytes) @ 0x00e4905b [0x00e48fd0+0x8b]
> J 1628 C1 org.opencv.core.Mat.finalize()V (12 bytes) @ 0x00e48df1 [0x00e48dc0+0x31]
> J 543 C1 java.lang.ref.Finalizer.runFinalizer(Lsun/misc/JavaLangAccess;)V (62 bytes) @ 0x00c40e64 [0x00c40ce0+0x184]
> J 1803% C1 java.lang.ref.Finalizer$FinalizerThread.run()V (55 bytes) @ 0x00c654bd [0x00c65220+0x29d]
> v  ~StubRoutines::call_stub
> V  [jvm.dll+0x158ba5]
> V  [jvm.dll+0x21e12e]
> V  [jvm.dll+0x158c3e]
> V  [jvm.dll+0x158dc6]
> V  [jvm.dll+0x158e37]
> V  [jvm.dll+0xfd84f]
> V  [jvm.dll+0x17c07c]
> V  [jvm.dll+0x17c16a]
> V  [jvm.dll+0x1c06f6]
> C  [MSVCR100.dll+0x5c556]
> C  [MSVCR100.dll+0x5c600]
> C  [kernel32.dll+0xb713]
> C  0x00000000
> 
> Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
> J 1629  org.opencv.core.Mat.n_delete(J)V (0 bytes) @ 0x00e49017 [0x00e48fd0+0x47]
> J 1628 C1 org.opencv.core.Mat.finalize()V (12 bytes) @ 0x00e48df1 [0x00e48dc0+0x31]
> J 543 C1 java.lang.ref.Finalizer.runFinalizer(Lsun/misc/JavaLangAccess;)V (62 bytes) @ 0x00c40e64 [0x00c40ce0+0x184]
> J 1803% C1 java.lang.ref.Finalizer$FinalizerThread.run()V (55 bytes) @ 0x00c654bd [0x00c65220+0x29d]
> v  ~StubRoutines::call_stub
> . . . . .
> 

- "error.log" ... timestamp matches start of the program in the afternoon (aborted using task manager, because the GUI hangs / camera  mounted on the machine and see how it looks like from within OpenPnP
> . . . . .
> Exception in thread "Thread-7" java.lang.OutOfMemoryError: Java heap space
> 	at java.awt.image.DataBufferByte.<init>(Unknown Source)
> 	at java.awt.image.Raster.createInterleavedRaster(Unknown Source)
> 	at java.awt.image.BufferedImage.<init>(Unknown Source)
> 	at org.openpnp.util.OpenCvUtils.toBufferedImage(OpenCvUtils.java:57)
> 	at org.openpnp.machine.reference.ReferenceCamera.transformImage(ReferenceCamera.java:207)
> 	at org.openpnp.machine.reference.camera.OpenCvCamera.capture(OpenCvCamera.java:75)
> 	at org.openpnp.machine.reference.camera.OpenCvCamera.run(OpenCvCamera.java:93)
> 	at java.lang.Thread.run(Unknown Source)
> . . . . .
>

vonnieda, 2016-03-25:
Thanks @ldpgh, it looks like this is probably caused by #243. I'm not sure why I don't see this issue on my development machine - it seems to be Windows specific. I'll look into this but it may be a few days before I have the chance.

ldpgh, 2016-03-25:
Btw. I do have a pyOpenCV installation in parallel. In addition there is an openCV installation (C:\Programme\opencv with timestamp Oct 2014). Maybe it's related to the pyOpenCV package.

vonnieda, 2016-03-25:
If you are able to remove any other traces of OpenCV from your system to test if that is the issue that would be really helpful, but I understand if that is asking too much. I think what is most likely is just that there is a Mat that's not being released properly and for reason on the 32 bit Windows VM that is more of a problem than on other VMs.

ldpgh, 2016-03-25:
- And to make it more complicated ... there is a regular Java in my search path
C:\> java -version
java version "1.6.0_25"
Java(TM) SE Runtime Environment (build 1.6.0_25-b06)
Java HotSpot(TM) Client VM (build 20.0-b11, mixed mode, sharing)

- Have renamed ".openpnp" to "openpnp_merken" and started openPnP again.
Tried "Webcams" instead "OpenCVCamera" ... executable is still running, but the Camera-Canvas is black (nothing to see) ... maybe it's fine.

- Renaming "C:\Programme\opencv" is not a big deal ... will try it with the OpenCV camera again.

ldpgh, 2016-03-25:
Back with OpenCvCamera ... video is runnign in the camera-canvas.
- 1GByte RAM still left.
- CPU running a full speed
- Have turned "Cameras" back to "Show None" ... CPU keeps busy, memory footprint stays unchanged (not counting up).

- GUI gone again ... exe still running. CPU is back to idle.

vonnieda, 2016-03-25:
Okay, thanks for the testing. I will try to reproduce on a clean VM and
that should at least tell us if it's an OpenPnP issue or an environmental
one. Probably be at least Sunday before I can take a look.

Jason

vonnieda, 2016-03-27:
@ldpgh Okay, I've been running OpenPnP on my Windows machine with the same setup you have in your machine.xml for the last 24 hours and no crashes. It does tend to consume a lot of memory over time but then eventually Java notices and garbage collects it all and the cycle continues.

So, I think there are two possibilities here:

1. You have a copy of opencv*.dll somewhere on your system that OpenPnP is picking up and that has a bug in it.
2. Your machine has low enough RAM that OpenPnP runs out of RAM before it notices that it should garbage collect. This is an issue with how OpenCV allocates memory but I think I can maybe improve the situation a bit by doing more extensive memory management with OpenCV.

So, my suggestion for the short term would be to see if you can remove any traces of OpenCV from your system and see if that fixes the problem. In the mean time, once I finish what I am currently working on I will see if I can make some memory management improvements. I will test on my Windows machine and see if I can keep the overall memory usage much lower over time, which should help.

vonnieda, 2016-03-27:
One more note for myself: Switching to loadLocal instead of loadShared for OpenPnP would resolve the problem of potentially using the wrong DLL.

ldpgh, 2016-03-29:
Have tried more  ...
1. Removed nearly all SET settings (e.g. CLASSPATH, PYTHONPATH, PATH, ...) in a new new cmd.exe window. Program has been started as "C:\openpnp\openpnp.exe". It gave me an uptime of about 20 minutes (compared with 3 minutes before). It stopped with HEAP  exception. Error log is available.
2. Have scannned my entire harddisc ... just the mentioned opencv directory, one opencv related file (USB microscope) found and renamed and the DLL files in the temp directory from the OpenPnP runs .. no improvement (just few minutes uptime).

Btw. a driver is still needed. At the moment a Python socket catches the output from the LinuxCNC driver and spills out tranlated commands to the COM port. So I could bring up the software/Java tooochain, but have no idea how to catch the runtime issue by the debugger or whatever is recommneded for Java developments. 
- What do you think?
- Have a Win7Pro Laptop already in my hand with no issues when running OpenPnP but I am not really using it (everything different, not better IMHO, partially ...). If you like to close this XP related issue it would encourage me to move to the new machine :-)

vonnieda, 2016-03-29:
Hi @ldpgh,

I'm going to run one more test using a 32 bit VM when I have a few spare
minutes. Hopefully that will show the issue. I'd like to fix it if I can
reproduce it. Either way, if you have a machine that OpenPnP works fine
with I heartily suggest you use that one instead :)

Jason

ldpgh, 2016-04-02:
Would like to support the testing ... need setup anyway, but first time use of Git :-o
- Have IntelliJ 2016.1 on both machines (XP & Win7), Git and repository cloned, Java/JDK 6,7,8 can be turned on
- Win7: repo cloned 29-Mar ... builds fine, exec runs as expected based on Java 8 :-)
- WinXP: repo cloned 1-Apr
   - Java 7 does not build JogControlsPanel.java due "language level not sufficient"
   - Java 8 setup complains on other files (jar files available, but ignored by javac)

What release matches the download from openpnp.org in the context of WinXP?
What is the right Java major version (6,7 or 8) in the context of WinXP?

ldpgh, 2016-04-02:
Test using ImageCamera, OpenPnP.exec from release package
- idle time = 2h 30min
- cpu time = 12min 35sec
Memory footprint stays flat since the beginning.

vonnieda, 2016-04-02:
Thanks @ldpgh, so just to be clear, it did not crash at all during that whole time?

ldpgh, 2016-04-03:
Your impression is right ... was running smooth, no issue at all.

vonnieda, 2016-04-04:
@ldpgh I've just checked in some changes in the develop branch that I hope will resolve this issue. Can you please download the latest and give it a try?

Unfortunately I still have not been able to reproduce the problem here, so I can't be sure this will fix it, but I did find a couple clear memory leaks in the camera handling code and fixed them, so I hope this will help.

ldpgh, 2016-04-04:
Update, build, run ... done

Unfortunately got a crash already with an uptime of about 25min (800MB free RAM).
There is (as far as I can tell) a significant improvement
- No more Java exceptions et all ... just the fatal Java Runtime issue is left.
- CPU load is just 37% (80% yesterday)

Have a Debug-Session started with Exception-breakpoint enabled. It's running for 60min already ... so far so good.

vonnieda, 2016-04-04:
Thanks @ldpgh. Can you verify that your machine.xml is the same as the one you posted previously? For instance, I want to make sure you don't have additional cameras, or that you have not applied lens calibration, rotation, offsetting or flip to the camera. These all introduce additional variables and I want to make sure we're staying focused on just the OpenCvCamera.

ldpgh, 2016-04-04:
It's definitely a different machine.xml ... now with GcodeDriver and my command subset.

vonnieda, 2016-04-04:
@ldpgh  Can you send the new one please? I just need to be sure we're not introducing additional complexity.

ldpgh, 2016-04-04:
Will provide the file ... need some seconds to get it done.

ldpgh, 2016-04-04:
please have look ... https://github.com/ldpgh/rasenmaeher/blob/master/machine.xml

ldpgh, 2016-04-05:
FYI ... did some testing today:

1)  NO aborts if using a debug-session with exception breakpoints turned on in IntelliJ

2.1)  1st program start in IntelliJ usually fails within 20 minutes
2.2)  2nd program start  NO  fails
2.3)  Using it in with controller connected (having its own issues) abort is back with an uptime of 20...40 minutes.

Strange and hard to debug.

ldpgh, 2016-04-05:
Java exception regarding heap memory right at start (about 800MB RAM free)

> 
java.lang.Exception: unknown exception
	at org.opencv.core.Mat.n_release(Native Method)
	at org.opencv.core.Mat.release(Mat.java:1893)
	at org.openpnp.machine.reference.camera.OpenCvCamera.capture(OpenCvCamera.java:80)
	at org.openpnp.machine.reference.camera.OpenCvCamera.run(OpenCvCamera.java:95)
	at java.lang.Thread.run(Thread.java:745)

>
Exception in thread "Thread-7" java.lang.OutOfMemoryError: Java heap space
	at java.awt.image.DataBufferByte.<init>(DataBufferByte.java:76)
	at java.awt.image.Raster.createInterleavedRaster(Raster.java:266)
	at java.awt.image.BufferedImage.<init>(BufferedImage.java:376)
	at org.openpnp.util.OpenCvUtils.toBufferedImage(OpenCvUtils.java:60)
	at org.openpnp.machine.reference.ReferenceCamera.transformImage(ReferenceCamera.java:205)
	at org.openpnp.machine.reference.camera.OpenCvCamera.capture(OpenCvCamera.java:74)
	at org.openpnp.machine.reference.camera.OpenCvCamera.run(OpenCvCamera.java:95)
	at java.lang.Thread.run(Thread.java:745)

ldpgh, 2016-04-05:
Java exception during run-time ... no abort at this time

> 22:52:23,005 DEBUG          GcodeDriver sendCommand(moveto  Y80.0000   F1000, 5000)...
22:52:23,396 DEBUG          GcodeDriver sendCommand(moveto  Y80.0000   F1000, 5000) => [ok]
22:52:24,257 DEBUG          GcodeDriver sendCommand(moveto  Y70.0000   F1000, 5000)...
22:52:24,678 DEBUG          GcodeDriver sendCommand(moveto  Y70.0000   F1000, 5000) => [ok]
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at org.openpnp.gui.MachineSetupPanel$3.valueChanged(MachineSetupPanel.java:141)
	at javax.swing.tree.DefaultTreeSelectionModel.fireValueChanged(DefaultTreeSelectionModel.java:635)
	at javax.swing.tree.DefaultTreeSelectionModel.notifyPathChange(DefaultTreeSelectionModel.java:1093)
	at javax.swing.tree.DefaultTreeSelectionModel.removeSelectionPaths(DefaultTreeSelectionModel.java:502)
	at javax.swing.JTree.removeDescendantSelectedPaths(JTree.java:3723)
	at javax.swing.JTree.setExpandedState(JTree.java:3606)
	at javax.swing.JTree.collapsePath(JTree.java:2237)
	at javax.swing.plaf.basic.BasicTreeUI.toggleExpandState(BasicTreeUI.java:2297)
	at javax.swing.plaf.basic.BasicTreeUI.handleExpandControlClick(BasicTreeUI.java:2273)
	at javax.swing.plaf.basic.BasicTreeUI.checkForClickInExpandControl(BasicTreeUI.java:2231)
	at javax.swing.plaf.basic.BasicTreeUI$Handler.handleSelection(BasicTreeUI.java:3600)
	at javax.swing.plaf.basic.BasicTreeUI$Handler.mousePressed(BasicTreeUI.java:3548)
	at java.awt.Component.processMouseEvent(Component.java:6532)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6300)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4891)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4522)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4713)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:331)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
23:03:23,765 DEBUG          GcodeDriver sendCommand(moveto  Y71.0000   F1000, 5000)...
23:03:24,136 DEBUG          GcodeDriver sendCommand(moveto  Y71.0000   F1000, 5000) => [ok]
23:03:26,970 DEBUG          GcodeDriver sendCommand(moveto  Y71.1000   F1000, 5000)...
23:03:27,320 DEBUG          GcodeDriver sendCommand(moveto  Y71.1000   F1000, 5000) => [ok]

ldpgh, 2016-04-07:
Status update ... started as debug-session in the IDE the uptime is 4:30h. Have restarted the programm just because openCV log-functionality has to be turned on.

ldpgh, 2016-04-20:
Since the last message just using IntelliJ/Debug rather than IntelliJ/Run. GUI is no longer vanished into thin air.
The IntelliJ setup is:
- Debug runs with "Java Exception Breakpoints / any exception" enabled.
- "Mute Breakpoints" turned on

vonnieda, 2016-11-27:
Have had no further reports of this. Seems like it was maybe IntelliJ related. Please reopen if this is still a problem.

closed 2016-11-27 by vonnieda