openpnp/openpnp #251: WinXP ... Temp directory access violation during clean up (remove)

ldpgh, 2016-04-02:
Context ...
- OS=WinXP, User="Admin"-Rights
- own compile using Intelly 2016.1 / WinXP / Java 8  ... but seems the same for public Win32 release download package
- Unpacked openCV data are not removed from the Temp-directory.
- But there is no issue with manual remove of the files/directory afterwards

In the IDE following stacktrace has been created while running and immediately leaving the executable (Exit)

C:\Programme\Java\jdk1.8.0_77\bin\java ...
12:42:17,432 INFO         Configuration No packages.xml found in configuration directory, loading defaults.
12:42:18,274 INFO         Configuration No parts.xml found in configuration directory, loading defaults.
12:42:18,304 INFO         Configuration No machine.xml found in configuration directory, loading defaults.
12:42:22,079 INFO         Configuration Defaults were loaded. Saving to configuration directory.
Exception in thread "Thread-6" java.lang.RuntimeException: java.nio.file.AccessDeniedException: C:\DOKUME~1\<username>\LOKALE~1\Temp\6669052286423873460\nu\pattern\opencv\windows\x86_32\opencv_java2411.dll
	at nu.pattern.OpenCV$TemporaryDirectory.delete(OpenCV.java:146)
	at nu.pattern.OpenCV$TemporaryDirectory$1.run(OpenCV.java:117)
Caused by: java.nio.file.AccessDeniedException: C:\DOKUME~1\<username>\LOKALE~1\Temp\6669052286423873460\nu\pattern\opencv\windows\x86_32\opencv_java2411.dll
	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:83)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.implDelete(WindowsFileSystemProvider.java:269)
	at sun.nio.fs.AbstractFileSystemProvider.deleteIfExists(AbstractFileSystemProvider.java:108)
	at java.nio.file.Files.deleteIfExists(Files.java:1165)
	at nu.pattern.OpenCV$TemporaryDirectory$2.visitFile(OpenCV.java:141)
	at nu.pattern.OpenCV$TemporaryDirectory$2.visitFile(OpenCV.java:130)
	at java.nio.file.Files.walkFileTree(Files.java:2670)
	at java.nio.file.Files.walkFileTree(Files.java:2742)
	at nu.pattern.OpenCV$TemporaryDirectory.delete(OpenCV.java:130)
	... 1 more

Process finished with exit code 0

vonnieda, 2016-04-02:
This is sort of a known issue in the OpenCV bindings we use. More information here: https://github.com/PatternConsulting/opencv/issues/10

There is a fix/hack here that might help address this: https://github.com/phrack/ShootOFF/issues/407

Leaving this open as we do need to come up with a fix for it in https://github.com/openpnp/opencv and then bring in the fixed version.

ldpgh, 2016-04-05:
Maybe ... 
(1) during session add an additional openpnp directory and store the current temp directory name
(2) next time on startup look in the know folder for old directories and do whatever is appropriate.

vonnieda, 2017-01-15:
This was fixed a while back in https://github.com/openpnp/opencv/pull/9. Due to the difficulty of actually removing the DLL on shutdown, now we instead just clean up any old ones we find on startup.

closed 2017-01-15 by vonnieda