openpnp/openpnp #254: StripFeeder autosetup does not return with the visual feedback

ldpgh, 2016-04-06:
According to the screenshot (next comment) I tried to use the Auto Setup function for a strip feeder as shown in movies
- https://www.youtube.com/watch?v=Fs-SwSq5AZw
- https://www.youtube.com/watch?v=n8ISop9YQUk

After pressing the button its title turns into "Cancel Auto Setup", but nothing else seems to happen. It looks something is moving forward as indicated by missing
- colored detection lines provided (I assume) by OpenCV as shown in the movie.
- text added to the images directing what the user has to do

ldpgh, 2016-04-06:
[image]

vonnieda, 2016-04-06:
@ldpgh Have you assigned your OpenCvCamera to the head? Last time you showed me your machine.xml it was not attached to a head, only the ImageCamera was.

ldpgh, 2016-04-06:
You are right. Have changed Head-field entries
- OpenCvCamera:Head = H1
- Zoom:Head = None

and missing the dialog/text entries appear.

It seems a restart was necessary, but I'm not sure at this point. In addition a 2nd resistors has been added to the strip below the one already there.

Unfortunately the processing returns with a dialog 
   Auto Setup Failure
   Index: 0, Size: 0
It does not tell me very much.

vonnieda, 2016-04-06:
Okay, next up: Have you set your Units Per Pixel setting for the
OpenCvCamera correctly? When you move the camera over the tape does the
camera view show any holes highlighted?

ldpgh, 2016-04-06:
You have perfect insight. Yes camera was not calibrated yet. Now it is. 

Auto Setup works and provides data.

Though the data are not satisfying at the moment due to a very wrinkled strip I think my report is   false_issue  an this issue can be closed.

Thank you Jason. Best regards, Lutz

closed 2016-04-06 by vonnieda