openpnp/openpnp #255: Remove Double.NaN from Driver workflows

vonnieda, 2016-04-06:
Currently the drivers are supposed to accept Double.NaN on a moveTo() to mean "Don't move this axis" but not all drivers handle it correctly, it's not documented very well and it's confusing for driver authors. 

It's only really used in code that hasn't been updated since the immutable Location refactor and since that refactor it's not really very useful since the majority of operations are happening on all four axes anyway.

I want to remove the Double.NaN behavior (after reviewing everywhere it's used) and add a check to Location to make sure that none of the values are NaN.

vonnieda, 2017-02-12:
Started this work in https://github.com/openpnp/openpnp/tree/feature/remove-nan

markmaker, 2021-04-25:
Handled by Advanced Motion Control.

closed 2021-04-25 by vonnieda