openpnp/openpnp #257: Job stops with "Bottom vision only implemented ..."

ldpgh, 2016-04-20:
While starting a Job at the first part a "PartError" message pops up with the message ... "Bottom vision only implemented for Up looking cameras".
Though OpenCV camera (looking DOWN) is attached and it's disabled ... running the job without using the vision system.
Could track the root cause down to  OpenCvVisionProvider.Java:235   with ImageCamera:"Up" ... 
[image]
but in the camera-tab image camera has property "Looking"="Down" assigned.
[image]

vonnieda, 2016-04-20:
The reason you are seeing this error is because you have a non-head attached camera (Zoom). It's a bug, since it should be ignored if it's not set to Up looking, but you can get around it just by removing that extra camera that you aren't using.

This bug is fixed in the new bottom vision code that will be released soon.

Note to self: Close this when #104 is complete.

vonnieda, 2016-04-25:
Fixed in #104.

closed 2016-04-25 by vonnieda