openpnp/openpnp #259: Job Step Should Do Full Placements, not Segments

vonnieda, 2016-04-23:
Currently when you step through a job the steps are very fine grained: Move head up, move head down, pick, place, move to placement, etc. This was very useful during early development of OpenPnP but is not particularly useful for users now. I think this should be chained to either medium grain or coarse grained steps. I'm open to opinions from users:

Option 1, Medium Grained: Steps would be:
1. Feed
2. Pick
3. Align
3. Place

Option 2, Coarse Grained: Each step would run a full placement cycle. Basically all of the steps in Option 1 as 1 operation. In other words, each time you click step a part gets placed.

I think that the second option is the best choice. I think the only time you'd want the first option is when setting up or debugging and there are separate buttons for each of those operations that you can already use without running a job.

richard-sim, 2016-04-26:
What about a hierarchy, using a tree control for the UI. That way, the user can expand the nodes if they want fine-grained stepping of actions, or collapse them and get progressively coarser-grained stepping.

I can see this also helping with #265, allowing for better handling of failures, repeating different operations (including previous operations that are associated with the failed one).

vonnieda, 2016-04-27:
@richard-sim I think that's probably too much flexibility. My feeling is that users want less complexity, not more.

vonnieda, 2016-05-14:
Fixed in #265.

closed 2016-05-14 by vonnieda