openpnp/openpnp #263: Improve icon design language

vonnieda, 2016-04-23:
When the current icon set was first designed I had planned to make it such that any icon that would result in the machine moving (which is potentially dangerous) would include red as it's primary color (to indicate danger) and no other icon would include red. I've drifted from this since then and I need to update the current icons and replace any that don't fit that standard. The various feed buttons and the nozzle change buttons come immediately to mind.

issus, 2016-08-08:
I'll take this one on, I'll hopefully get time to start looking at it next week.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda