openpnp/openpnp #267: Improved Jog Controls

vonnieda, 2016-04-30:
The current jog controls are a little clunky to use because you have to set the increment to move different distances. We can improve this by integrating the distances into the controls. 

Pronterface and LaserWeb both use the same widget, which is very nice. I believe it originally comes from Pronterface. We can either check with them to see if they wouldn't mind if we copied it completely or we can develop one of our own with similar traits.

Example here:
https://github.com/openhardwarecoza/LaserWeb#machine-control-widget

SVG for it here:
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L443

TomKeddie, 2016-05-25:
Since both are gplv3 should be fine.  Images are at [XY](https://github.com/kliment/Printrun/blob/master/images/control_xy.svg) and [Z] (https://github.com/kliment/Printrun/blob/master/images/control_z.svg).

!(https://raw.githubusercontent.com/kliment/Printrun/master/images/control_z.svg)

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda