openpnp/openpnp #268: Window unusably small on 2560x1600 Retina Display

vonnieda, 2016-05-03:
See thread at: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/tJDUyj3xt_Y/x_AfiWitKwAJ

[image]

I'm not exactly sure what the right way to fix this is. Going to require some research. If anyone has any experience with Java and Retina displays, please speak up.

vonnieda, 2016-05-04:
Some useful links on this subject:
http://bulenkov.com/2013/06/23/retina-support-in-oracle-jdk-1-7/
http://stefan.budeanu.com/blog/?p=88
http://stackoverflow.com/questions/20315371/retina-compatible-jlabel-image

vonnieda, 2016-05-04:
Honing in on setting font size:
http://stackoverflow.com/questions/7434845/setting-the-default-font-of-swing-program

richard-sim, 2016-05-19:
Perhaps the Machine Controls panel needs to have a vertical scrollbar, so all the controls will remain accessible (but may require a bit of scrolling) while the bottom panel can be any height. As you've found, technically the behaviour of Java is correct as-is; just the window layout doesn't work for low-res screens (which that Retina resolution essentially is!).

Doing this, the minimum height of the Machine Controls panel can be removed, and the user can expand the bottom panel substantially even on low-resolution screens.

vonnieda, 2016-05-20:
Thanks @richard-sim, this is helpful. I think adding a scrollbar would be a good short term solution. Long term I'd like the window layout to work a little better for these screens.

tracklogic, 2016-08-03:
Scrollbar also sounds sensible incase user tries to run on a tiny 10" Netbook!

vonnieda, 2016-12-21:
With the changes in https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2016-12-20 I think we can call this one done.

closed 2016-12-21 by vonnieda