openpnp/openpnp #273: Zero after homing

psykhon, 2016-05-21:
The software wont set its DROs to zero after homing
[https://goo.gl/photos/He8T7zLD9LSx5xq86](url)

`
<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine" speed="1.0">
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="22964dce-252a-453e-8106-65db104a0763" name="H1">
            <nozzles>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="69edd567-df6c-495a-9b30-2fcbf5c9742f" name="N1" pick-dwell-milliseconds="0" place-dwell-milliseconds="0" current-nozzle-tip-id="e092921a-2eef-449b-b340-aa3f40d8d791" changer-enabled="false" limit-rotation="true">
                  <nozzle-tips>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d791" name="NT1" allow-incompatible-packages="true">
                        <compatible-package-ids class="java.util.HashSet"/>
                        <changer-start-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <calibration angle-increment="15.0" enabled="false">
                           <pipeline>
                              <stages>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="9" enabled="true" settle-first="true"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Bgr2Gray"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.BlurMedian" name="4" enabled="true" kernel-size="3"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="3" enabled="true" diameter="200"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="2" enabled="true" threshold="190" auto="true" invert="false"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="5" enabled="true" diameter="100"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="result" enabled="true"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="0" enabled="true" image-stage-name="9"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="7" enabled="true" key-points-stage-name="result">
                                    <color r="255" g="0" b="204" a="255"/>
                                 </cv-stage>
                              </stages>
                           </pipeline>
                        </calibration>
                     </nozzle-tip>
                  </nozzle-tips>
                  <head-offsets units="Millimeters" x="-72.5" y="10.6" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </nozzle>
            </nozzles>
            <actuators>
               <actuator class="org.openpnp.machine.reference.ReferenceActuator" id="5b8a8cbd-d6c1-4324-af8b-eba5f0444622" name="A1" index="0">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </actuator>
            </actuators>
            <cameras>
               <camera class="org.openpnp.machine.reference.camera.OpenCvCamera" id="34471928-02e5-4d3f-bc14-a308b821e6d3" name="OpenCvCamera" looking="Down" settle-time-ms="250" rotation="0.0" flip-x="false" flip-y="false" offset-x="0" offset-y="0" crop-width="0" crop-height="0" device-index="0" preferred-width="0" preferred-height="0" fps="24">
                  <units-per-pixel units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <vision-provider class="org.openpnp.machine.reference.vision.OpenCvVisionProvider"/>
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
                  <calibration enabled="false">
                     <camera-matrix length="9">0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.413989845E-315, 2.4139902E-315, 9.486E-321</camera-matrix>
                     <distortion-coefficients length="5">0.0, 0.0, 0.0, 0.0, 0.0</distortion-coefficients>
                  </calibration>
               </camera>
            </cameras>
            <paste-dispensers>
               <paste-dispenser class="org.openpnp.machine.reference.ReferencePasteDispenser" id="53050ccf-59a0-4d9f-a8d3-6216f5412e4e" name="D1">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </paste-dispenser>
            </paste-dispensers>
            <park-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
         </head>
      </heads>
      <feeders>
         <feeder class="org.openpnp.machine.reference.feeder.ReferenceStripFeeder" id="923eedf1-6a1a-4bbc-b603-421066262adc" name="R0805" enabled="true" part-id="R0805-1K" retry-count="3" tape-type="WhitePaper" vision-enabled="true" feed-count="0">
            <location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            <reference-hole-location units="Millimeters" x="147.347" y="40.285" z="0.0" rotation="0.0"/>
[... 231 lines cut ...]
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Bgr2HsvFull"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.MaskHsv" name="2" enabled="true" hue-min="60" hue-max="130" saturation-min="0" saturation-max="255" value-min="0" value-max="255"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="3" enabled="true" conversion="Hsv2BgrFull"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="6" enabled="true" conversion="Bgr2Gray"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="12" enabled="true" threshold="100" auto="false" invert="false"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.FindContours" name="5" enabled="true" retrieval-mode="List" approximation-method="None"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.FilterContours" name="9" enabled="true" contours-stage-name="5" min-area="50.0" max-area="900000.0"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="11" enabled="true" diameter="0"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.DrawContours" name="7" enabled="true" contours-stage-name="9" thickness="2" index="-1">
                           <color r="255" g="255" b="255" a="255"/>
                        </cv-stage>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.MinAreaRect" name="result" enabled="true" threshold-min="100" threshold-max="255"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="14" enabled="true" image-stage-name="0"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.DrawRotatedRects" name="8" enabled="true" rotated-rects-stage-name="result" thickness="2"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="15" enabled="true" prefix="bv_result_" suffix=".png"/>
                     </stages>
                  </pipeline>
               </part-settings>
            </entry>
         </part-settings-by-part-id>
      </part-alignment>
      <fiducial-locator class="org.openpnp.machine.reference.vision.ReferenceFiducialLocator"/>
   </machine>
</openpnp-machine>
`

psykhon, 2016-05-21:
On a further review, seems like the Parking button is sending the noozle offset to the machine instead of the Head 0. Dont know if this is a misconfiguration of mine or a real bug

vonnieda, 2016-05-23:
This is a bug in the GcodeDriver. It needs to send a head movement update after homing.

@psykhon Park moves the selected nozzle to the Park location, which you can define in Machine Setup -> ReferenceMachine.

vonnieda, 2016-05-26:
Added a new issue for the Park specific part of this: https://github.com/openpnp/openpnp/issues/279

vonnieda, 2016-11-06:
This was fixed quite a while back in the ReferenceHead.

closed 2016-11-06 by vonnieda