openpnp/openpnp #274: Visual Homing

vonnieda, 2016-05-23:
To increase the accuracy of home location, we should have visual homing after mechanical homing.

See discussion at https://groups.google.com/d/msgid/openpnp/ecc9b101-e26c-4a60-8ec1-17d9cba83958%40googlegroups.com?utm_medium=email&utm_source=footer

mrandt, 2016-07-28:
As you're probably aware of, this is solved in LitePlacer software by putting a "fiducial" marker on the table, which the camera homes onto after mechanical homing. 

I think it would be beneficial if user could provide a custom video processing pipeline for the detection of this mark so users would have flexibility in choice of black on white, white on black, hole in metal, etc. depending on the table top surface.

tracklogic, 2016-08-02:
Implemented on branch, fiducial homing, will initiate a pull request to get into main

vonnieda, 2016-08-27:
@tracklogic I just noticed that this throws an exception if no homing fiducial is defined. I thought it would just silently ignore it. I'm changing the code now so that it does, since this breaks anyone not using fiducial homing. Did you intend for this, or was there some other plan?

vonnieda, 2016-08-27:
@tracklogic Fixed above in https://github.com/openpnp/openpnp/commit/833987580c373ac8d8178ffba37924cf5ee7af0c, please let me know if you have any thoughts on whether this should act differently.

tracklogic, 2016-08-27:
> @tracklogic <https://github.com/tracklogic> Fixed above in 8339875

Hi Jason

The intention was if none was defined, it wouldn't do the visual fiducial
homing - my apologies!

Matt

vonnieda, 2016-10-28:
Fixed in previous PR.

vonnieda, 2016-10-28:
Reopening until this feature is documented.

vonnieda, 2016-11-27:
Documented at https://github.com/openpnp/openpnp/wiki/GcodeDriver#visual-homing

closed 2016-11-27 by vonnieda