openpnp/openpnp #278: Stopping a job should stop the job as soon as possible

vonnieda, 2016-05-26:
With the new JobProcessor refactor the Stop button is less useful than it was. It now signals the JobProcessor to stop after the current cycle finishes, and this may mean quite a few more operations.

Instead, the Stop button should stop the job immediately (after the current operation completes) and then perform the cleanup cycle.

pk7639, 2017-06-09:
The same applies to the pause function. We commonly use pause on other machines to freeze the job  to trouble shoot things like a pick position error. At present the job processor will continue through to the end of the place operation before stopping.

vonnieda, 2019-02-21:
This seems to be due to FSM processing in a while loop, and it seems like that should not be the case since already call next() in a loop, but removing it breaks the job at the cleanup stage. Still investigating.

vonnieda, 2019-02-21:
The core issue is that the FSM loops while there is a nextMessage. I think the nextMessages are supposed to group "invisible" steps so it doesn't look like clicking step did nothing.

commit ca3d5426 (2019-02-21): Fixes #278. The JobProcessor now has an opportunity to pause or stop between every step aside from "do nothing" steps. This means that if you start the job and immediately hit pause it will complete the fiducial checks and then stop, rather than perform a full cycle.
Note that this change increases the granularity of the step function so that now stepping takes more clicks to do a full cycle, but I think this is a good tradeoff without having to rewrite the entire processor.
commit 545d9aa0 (2019-02-22): Update changes for recent changes.
Fixes #478.
Fixes #278.
commit c0b5b012 (2019-06-01): Merge branch 'develop' into test

* develop: (103 commits)
  Added a bare minimum pipeline example.
  Fixes override sounds not working for SoundSignaler. Added configuration wizard for SoundSignaler.
  Fixes #814. Home action's icon is now tied to a property which is bound to the machine's "homed" property. Added a few notes for future improvements in the same vein.
  Fix a minor issue in the test driver where getLocation did not apply head offsets. Fix some lint in the ReferenceHead which caused tests failures when the driver was updated at runtime.
  Improved documentation on new planner.
  [... cut]
closed 2019-02-21 by vonnieda