openpnp/openpnp #279: Park Improvements

vonnieda, 2016-05-26:
Originally from #273, some improvements to the Park system are needed:

* Park should be a method on head rather than being a bunch of copypasted code everywhere. One thing to consider is that this would break the UI which allows parking of XY, C or Z. Maybe the method would take those as booleans.
* Park should always park at the same location by subtracting the offsets of the object it moves. Due to the driver API we can't just send an absolute coordinate, but we can pre-subtract the head offsets from the default nozzle, for instance, and send that.

cri-s, 2017-04-05:
Take care for machines having more then one head, and for that park can issue a exception if the move is not allowed because of position from other head.

vonnieda, 2017-04-05:
@cri-s Do you know of any machines currently using OpenPnP with more than one head?

closed 2019-02-13 by vonnieda