openpnp/openpnp #280: JobProcessor Improvements v2

vonnieda, 2016-05-26:
This is a grouping ticket for several issues related to the recent JobProcessor refactor:

* [ ] Stopping a job should stop the job as soon as possible #278
* [ ] New JobProcessor needs better recovery options #275
* [ ] Solder Paste Dispense Broken After JobProcessor Refactor #271

cri-s, 2016-07-02:
Annother JobProcessor issue inside FSM , no defined transition from Cleanup to Initialize .
What is the reason ?

vonnieda, 2016-07-03:
@cri-s Are you asking why there is not a transition defined for this, or did you receive this as an error?

cri-s, 2016-07-03:
Im receive this as error.

vonnieda, 2016-07-03:
Please file a new bug report and include information on how it happened

cri-s, 2016-07-03:
Basically new install, added driver for grbl, changed dragFeeder
(added optional delay and back movement before going up)  code, added
light actor handling inside BottomVision for camera.
When starting job, that error came up. some idea what is the reason ?

cri-s, 2016-07-03:
done, #295

vonnieda, 2017-08-09:
Note, per discussion in https://groups.google.com/d/msgid/openpnp/4b284469-fb53-4e17-a5f4-45b696c23f7a%40googlegroups.com there is a bug where discard might discard for the wrong nozzle. I think what is happening is that skip is not taking the right planned placement into account.

KarlZeilhofer, 2017-08-15:
Hi, I just crashed the nozzle into a 2.5mm high chip, while placing a C0805 next to that chip. The job processor seam to decided to place the big chip first, because the big nozzle was inserted. After that the small parts got placed, an the crash occurred. 

Is there any chance for the user to see, what is going to happen?

vonnieda, 2017-08-15:
@KarlZeilhofer Sorry to hear that. No, there is currently no way for the user to see or modify the order of placements. That's an important part of this issue's work.

mgrl, 2017-12-26:
hi @vonnieda, 
I just made a patch to have in a multinozzle-setup the right nozzle cleared if skipping a part due to errors: #693

markmaker, 2018-07-30:
Referring to the point...

> Actions: Plan (Reorders According to Planner)

...is this rework in any way related to #686 ?

If so, some thoughts about this: 

1. Ideally the Planner would be a separate object/class that one can choose in a Drop-Down much like the machine driver. 

The Planner should then have a simple API, mostly just being notified when it is time to do the planning. 

The Planner should in turn have access to the full job and machine data, because it may vary wildly from machine to machine by _which_ criteria the Planner will order/group the placements.  A multi-nozzle machine might want to order/group parts by feeder proximity to reduce moving time between pickups. Conversely a single nozzle machine user doesn't care about feeder position and might favour assembly by PCB X or Y. Another [retrofitted] machine with small Z headroom might require strict part height sorting. Yet another one might want to implement [this crazy idea](https://makr.zone/planning-the-work-area/117/) :-).  

See also:
https://youtu.be/y_Fp3wLdmtk?t=1m19s

Perhaps the rework could also include the following considerations:

2. The placement could include states "not yet planned", "auto-planned", "manually planned" to allow for gradual determination of the placement ordinal. 
3. Planning could then be incremental, the Planner may choose to implement a lazy evaluation and/or background and/or iterative optimization scheme. 
4. There may be dominant order/grouping properties (such as similar part heights) pre-determining a sub-selection of candidate placements for lazy evaluation. This gradual (hierarchic) planning may offload the CPU on large jobs and/or complex optimization criteria.  
5. Iterative planning may allow for algorithms that gradually improve the placement solution (i.e. [Simulated Annealing](https://en.wikipedia.org/wiki/Simulated_annealing)) without unduly slowing down job start/failure recovery (i.e. after a feeder fails). The Planner may revisit placement ordinals which are "auto-planned", but not those that are "manually planned".
6. The Planner should be notified when various events happen (job started, paused, resumed, next placements needed, feeder failed, user changed data, idle, etc.) but it would be up to the Planner implementation, _when_ the actual planning (i.e. the determination of the placement ordinal) takes place. 

_Mark

vonnieda, 2018-07-31:
Hi @markmaker,

Yes, the two are certainly related.

OpenPnP used to have a separate planner interface. I removed it because it was complex, hard to maintain, and did not fit well with the first major refactoring of the JobProcessor. And I also didn't think anyone was really using it.

One thing I'm struggling with is the competing requests of:

1. Users want to be able to manage the order of placements.
2. Developers want a planning interface.

I have been thinking that the order of placements might just be defined by the order in the table, and that the planner would be something you can run as an action to set the order of placements.

I would say that there have been far more requests for the ability to modify the placement order than there have been for a custom planner. I'd be interested in ideas on how to manage both.

Finally, I should note that the addition of a new planning system will probably be secondary to the JobProcessor refactor. My primary goal with the refactor is to solve some major bugs, and to fix some major usability issues. Adding the planner later on should not really be much more additional work since the area where planning happens is already quite self contained.

Jason

vonnieda, 2018-07-31:
@markmaker See https://github.com/openpnp/openpnp/issues/219 for some more information and discussion on this.

markmaker, 2018-07-31:
Hi Jason

thanks for your response. 

> One thing I'm struggling with is the competing requests of:
>   1. Users want to be able to manage the order of placements.
>   2. Developers want a planning interface.

Yes that's why I asked for the implementation to include states "not yet planned", "auto-planned", "manually planned". This way both demands could be met and even mixed. 

This can be implemented very simply, say ordinal unplanned=0, auto-planned < 0, manually planned >0. The JobProcessor would then take the abs(ordinal) to sort the placements.

**The only thing I ask for at this point is for the ordinal handling/storing to support such a thing. The actual Planner on top of it can wait.**

Why the states?

The idea is that the [future] planner is free to change the ordinal <= 0 on all placements but **not** those > 0 determined by the user. The best analogue that comes to my mind is auto-routing with [Fritzing](http://fritzing.org/home/). Not that I use Fritzing nowadays but this was a very nice experience.  

_Mark

vonnieda, 2019-05-27:
Most of this is now fixed in https://github.com/openpnp/openpnp/commit/97e5bfc54827965a24eb710fa0932c3cf46ce29f. I'm closing this issue and will open new issues for the parts that remain. Much of the above is outdated with the new work, so I think it's best to start fresh. If https://github.com/openpnp/openpnp/commit/97e5bfc54827965a24eb710fa0932c3cf46ce29f does not address something you were concerned with in this issue, please file a new one.

closed 2019-05-27 by vonnieda