openpnp/openpnp #281: Pendant support for manual head operation

TomKeddie, 2016-05-31:
I am wondering if OpenPNP supports MPG pendants ? I am thinking it would be helpful for setup just like it is on a CNC mill .....

vonnieda, 2016-05-31:
Discussion at https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/hFp_RPG6tks/9Lbl3mtZAQAJ

As noted in the discussion, pendants are not currently supported.

vonnieda, 2016-06-01:
Per the discussion, I think we'd be best off handling this outside of OpenPnP using either keyboard emulation hardware or software. We may need to add a couple new shortcuts to handle things like absolute jog increments. The existing shortcuts are listed at https://github.com/openpnp/openpnp/wiki/User-Manual#shortcuts.

TomKeddie, 2016-06-14:
I have one of the joystick shields shown below that I intend to hook up to an Arduino Leonardo and do keyboard emulation - there's enough controls for X, Y and Z plus step size change.  There's no indication of step size unfortunately, might hack in some leds.

[image]

vonnieda, 2016-12-19:
Based on discussion, it was decided to do this outside of OpenPnP via keyboard controls.

closed 2016-12-19 by vonnieda