openpnp/openpnp #282: Scripting language support

TomKeddie, 2016-05-31:
One way of avoiding lots of enhancements is to include a scripting language where enhancements can be tried out before they are migrated to features.  Examples would the the ULP language in Eagle or ruby in Sketchup.  Some kind of menu that lists the scripts for execution would be needed.

Examples of features I would like to try in this environment would be.
- reset all feed counts of all strip feeders to zero
- adjust the X or Y location of all strip feeders when a rack is replaced in slightly different location
- adjust all Z locations when a head is changed
- remove all unused parts
- remove all unused packages

Much of this could be done offline manipulating the XML but could be smoother and more flexible integrated the system.

I don't have any examples of languages used like this in Java, examples would help in the choice of language.  Some random choices are python, ruby, tcl, java and javascript.

vonnieda, 2016-05-31:
Reference material:
http://docs.oracle.com/javase/6/docs/technotes/guides/scripting/programmer_guide/
http://www.java2s.com/Tutorial/Java/0120__Development/RunningScriptswithJavaScriptEngine.htm
http://www.java2s.com/Tutorial/Java/0120__Development/ListingAllScriptEngines.htm

vonnieda, 2016-05-31:
Discussion at: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/8la-W2aBxNY/4PxhuuyJAQAJ

vonnieda, 2016-06-01:
I think this would be a useful feature, and relatively easy to implement. As linked above, Java has pretty solid built in support for scripting languages and even ships with JavaScript built in. Many other languages can be added just by including the jar in your classpath.

Basic set of tasks to implement this:
* Add a new Scripts top level menu that lists all of the scripts in the .openpnp/scripts directory.
* Add an item to the bottom of the menu to rescan the directory - or just rescan every 10 seconds or so.
* Add an item to the bottom of the menu to open the scripts directory with the OS's built in file explorer.
* Expose, at least, the Configuration object to the script globals: This gives access to the entire machine, the feeders, the cameras, etc.
* Expose MainFrame to the script globals: This gives access to most of the UI.

ldpgh, 2016-06-07:
Just my 5 cents regarding Embedded interpreter (by intention prevent use of scripting language).

My personal preference is Python (with more or less experience over the years with Tcl, Python, Lisp). Though not the 1st candidate in the context of Java it allows to have a common system for Embedded use as well as the ability to use it outsid OpenPnp for communication purpose. On my machine it's glues the OpenPnP/GCodeDriver output to my machine including the command translation as well as adding the asynchronous sync signal (aka sending "ok" to the host).

As far as I understand (collected from discussions) the use of Javascript as a commandline interpreter is not a simple task ... maybe I'm wrong.

Have used SWIG to wrap my C++ stuff on the fly at every compile run (but just 2000 lines of code, about 30 files). For me (as a weekend programmer) it was suffient enough. No idea how good/easy it is for JAVA.

Btw. a good examples for Python integration is FreeCAD.
AFAIK Blender uses Python as well.
Have been told OpenOffice have Python scripting as well.

Just for my curiosity ... what programs/tools use Javascript as Script-engine the user allows to manipulate the main program easily.

And not to forget ... it comes with PDB (Python Debugger).
Instrumentation (code injection duing run-time) as it is possible in Java is possible in Python as well.

anthonywebb, 2016-06-14:
Javascript is ubiquitous, I would start there

vonnieda, 2016-06-20:
I have this just about done, so don't start on it :)

vonnieda, 2016-06-21:
Finished. See https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2016-06-20

closed 2016-06-21 by vonnieda