openpnp/openpnp #283: StripFeeder - partPitch collides with inaccuracy of strips

ldpgh, 2016-06-07:
At least the strips I'm using have a cummulative error over many part locations along the strip. I need to manually adjust the value from 4.00 to 3.986 to get parts picked reliable above position 10.
- Next hole postion is the postion of part number 30 in the strip. So the dX, dY values are there.
- dY is used
- dY uses the unmodifed partPitch value in ReferenceStripFeeder.java

vonnieda, 2016-06-07:
@ldpgh Are you using vision? The vision function is intended to correct this drift over time.

ldpgh, 2016-06-07:
Using un-modified partPitch of 4.00mm
[image]

ldpgh, 2016-06-07:
Using partPitch of 3.986mm
[image]

vonnieda, 2016-06-07:
@ldpgh Thanks for showing that, but I am confused on what you are actually asking for. It seems it's working fine if you adjust the Part Pitch value, and that you are not using Vision ("Use Vision?" is turned off).

If you are suggesting a change to the software, can you explain what you want to see changed? And have you tried using the Vision feature? It will automatically correct for this error.

ldpgh, 2016-06-07:
Using partPitch of 4.00mm modified in ReferenceStripFeeder.java
[image]

ldpgh, 2016-06-07:
Just to have all pieces at the same location (circumvent Git due to missing experience.) ...

Modified file ReferenceStripFeeder.java ... 
[ReferenceStripFeeder.java.txt](https://github.com/openpnp/openpnp/files/303136/ReferenceStripFeeder.java.txt)

ldpgh, 2016-06-07:
Changes made ...
line_48 ... added  "import static java.lang.Math.round;"
line_141-160 ... changes including comment

Added comment in the file is:
        // partPitchAdjusted is derived from the distance of ReferenceHole and NextHole and divided
        // by the amount of part location in between ... part count is calculated base on the distance
        // and the given partPitch
        // partPitch limitation ... It's the P1 value according to EIA-481-C, October 2003, pg. 9, 11, 13
        // without taken into account the variations as specified in this document.

ldpgh, 2016-06-07:
@vonnieda
The visions system moves slightly away over the distance and looses the focus. The root cause is not fully understood and could be affected by my feeders. There are blocked holes in the strip to hold it firmely at the measured position. Maybe I need to find a better solution.

ldpgh, 2016-06-07:
@vonnieda 
Changing the value of partPitch according to the strip accuracy is what I try to prevent.

To my understanding it's a value according to the  EIA spec. and for this reason I think it should stay untouched as an basic input parameter for the strip. Deriving manually the value from both holes (reference and next) ist tedious.

With next hole beeing any hole near the end of the strip it seems to me the partPatchAdjusted can be easily calculated according the measured locations

The impact of the proposed change to the vision system is not investigated and may require a a reject. Any decision is fine for me.

At the end the weekend programmer need to say
- I'm aware of https://github.com/openpnp/openpnp/wiki/Developers-Guide ... still ramping up the development setup
   - pnp-test.xml + unchanged source code = ok
   - pnp-test.xml + changed source code = FAIL ... (after Fiducial-tests, Placement coordinates have infinite values for X and Y)
- There is no automated test created.
- The change is in use and the behavior is constantly obeyed

ldpgh, 2016-06-07:
Just for completeness logfiles of pnp-test.xml with timestamps and camera-IDs removed to allow file-diff
- [OpenPnP.log.ok.txt](https://github.com/openpnp/openpnp/files/303314/OpenPnP.log.ok.txt)
   - many surprising NaN occur in moveTo()
   - 1st issue at line 1090 ... NaN rather than coordinates in moveTo()
   - 2nd issue at line 1146 ... different coordinates between OK and FAIL run in moveTo()
   - 3rd issue at line 1152/1153 ... Infinite in moveTo() let the program abort
- [OpenPnP.log.failed.txt](https://github.com/openpnp/openpnp/files/303319/OpenPnP.log.failed.txt)
   - many surprising NaN occur in moveTo()

vonnieda, 2016-06-07:
Thanks @ldpgh. This makes sense and I will look into fixing it.

The moves to NaN are a quirk of the driver model. NaN just means "don't
move this axis in this move". More information about that here:
https://github.com/openpnp/openpnp/issues/255

Jason

ldpgh, 2016-06-07:
RootCause for failing pnp-test.xml ... 
- just X-axis has been considered in the upload source code. If the strip is along the Y-axis the partPitch calculation is unrealistic small and the calculated distance is wrong

Test using calculation of the euclidian distance by Location::getXyzDistanceTo().
- previous ... `double partPitchAdjusted = lineLocations[0].getX()-lineLocations[1].getX();`
- new ... `double partPitchAdjusted = lineLocations[0].getXyzDistanceTo(lineLocations[1]);`
- new ... pnp-test.xml + changed source code = **OK**

@vonnieda 
It seems to me there a couple of distances slightly different to the ideal value (at least 2 were seen) in the pnp-test.xml setup. Would be good if you can confirm this.

ldpgh, 2016-06-08:
Update File [ReferenceStripFeeder.java.txt](https://github.com/openpnp/openpnp/files/304967/ReferenceStripFeeder.java.txt)
- comments polished, typos fixed
- reduce number of changed source code lines

Changes made ...
- line_48 ... add "import static java.lang.Math.round;"
- line_141-148 ... add comments
- line 149-150 ... add double variable `partPitchAdjusted`
- line_152 ... replace `partPitch.getValue()` by `partPitchAdjusted` in the expression

Running test job `pnp-test`
- Job `pnp-test` + changed source code = OK ... completed normally / No issues seen.

vonnieda, 2016-06-15:
Thanks @ldpgh. I have merged this code in commit 21804194c14a5b19f37666ce6613685637e3827a.

closed 2016-06-15 by vonnieda