openpnp/openpnp #284: error saving config

psykhon, 2016-06-09:
[https://drive.google.com/file/d/0B-ZJ2SAKAbyRMEhTTGpnODB4Ums/view?usp=sharing](url)

vonnieda, 2016-06-09:
@psykhon This url is not valid. If it's for a screenshot or an image can you just drag and drop it into a comment here please?

vonnieda, 2016-06-09:
@psykhon Nevermind previous comment, got the image.

[image]

vonnieda, 2016-06-09:
@psykhon It looks like you have the bottom vision pipeline for either a part or for the default and you've left the prefix blank on an ImageWriteDebug stage. If you find the one you modified you can just put any string there, or delete that stage and it should save.

closed 2016-06-13 by vonnieda