openpnp/openpnp #288: Logging Window or Tab

vonnieda, 2016-06-13:
Suggestion from @TomKeddie:

We should have a logging window or tab that can be shown or hidden from the view menu. This will be very helpful for users setting up new machines and helping with debugging. It will also be very important for the scripting feature #282.

* Required: Add a new Log tab to the main tabs interface.
* Required: Make it show/hide from the View menu.
* Nice to have: Add checkboxes or a dropdown menu in the tab to turn on and off stdout, stderr and common logging situations, e.g. driver, vision, gcode, etc.
* Nice to have: Add a search box.
* Nice to have: Click the class name of a log message to adjust the log level for that class.

TomKeddie, 2016-06-13:
Nice to have: save to file.
Nice to have: copy to clipboard/paste buffer.

vonnieda, 2016-06-17:
First pass at this is in.

dzach, 2017-01-10:
Provide a way to clear the log widget (not the log file) and show only new entries from that point in time on. Could be a filter for time (i.e. filter out entries earlier than hh:mm:ss) with a button to set time to "now"

pfried, 2017-01-23:
I already fiddled around with the logging on the weekend for some hours, but i had no luck in redirecting the output stream (stdout, stderr) to the Logger and keeping the output on the console. Can anyone give a hint on this or has a link to a working implementation of such a logger redirect?

vonnieda, 2017-01-23:
@pfried 
Pseudocode:
PrintStream originalOut = System.out;
LoggingPrintStream out = new LoggingPrintStream(originalOut);
System.setOut(out);

The basic idea is to make a new implementation of PrintStream that sends it's output both to the original System.out and to the Logger. The same can be done for err with setErr.

pfried, 2017-01-23:
okay, same approach i had, but no luck. I cant remember the problem correctly since it was already some weeks ago. I will try again.

pfried, 2017-03-09:
just wanted to leave a quick working notice

I started working on the features above, some things to note:

- I will replace the simple TextOutput by custom elements for each Logentry based on  a JList
- Output of the Logging to stdout will be back by adding a ConsoleLogger similar to the Tinylog one but with fixed original output PrintStream

About the UI:

For the search box and also for filtering i think i will "live" filter the underlying LogEntry List. I am thinking about the same for the LogLevels. The LineLimit is a bit pointless then, i would rather remove the line limit and set it to a fixed number.

All other functionalities will happen on the filtered list e.g. save to file, copy to clipboard

pfried, 2017-03-12:
About the issue:

> Nice to have: Add checkboxes or a dropdown menu in the tab to turn on and off stdout, stderr and common logging situations, e.g. driver, vision, gcode, etc.

I added one for the System.out, but not for others, additional checkboxes are very easy to add if necessary

>  Nice to have: Add a search box.

Done, searches whole LogEntry, so searching for class name is ease, also debug levels etc.

>  Nice to have: Click the class name of a log message to adjust the log level for that class.

I am not so sure if that makes sense, i did not do it because all the permanent settings are hard to display, you would need a list for all logging classes (or the relevant ones) with their corresponding log levels

>  Nice to have: Detach the tab to a floating window so the user can see the log messages while working in other tabs.

Acutally i do not know how to do this, maybe someone who did the multi window thing should take care of it.

>  Nice to have: Save to file.

Did not add as i think copying to clipboard is way easier and more helpful. Logs get saved anyway

>  Nice to have: Copy to clipboard/paste buffer.

Added a button for that

vonnieda, 2017-03-13:
Agree on all points.

vonnieda, 2017-03-16:
Example of output with extraneous newlines:
[image]

vonnieda, 2017-03-16:
Many thanks to @pfried for finishing up many of the requested features!

vonnieda, 2017-03-20:
Some of the features have been temporarily reverted due to a performance problem in #467.

pfried, 2017-03-20:
i saw it, currently i am running a profile run, @vonnieda could you test disabling autoscroll and check if the problem persists?

vonnieda, 2017-03-20:
@pfried  Yes, disabling auto scroll seems to fix it.

pfried, 2017-03-20:
About the general workflow, i think it would actually be nice if we had a more feature branch based workflow where features are in their own branch getting merged at a certain point. Of course there would need to be people who test the branches, but having to hotfix the develop branch is no solution either.

I now cherry picked the commits related to logging and reapplied them. The history now has a ton of entriesm

vonnieda, 2017-03-20:
@pfried re: feature branches: this is how I do all of my work, and there's no reason that others can't, too. When I start a new feature I create a branch called feature/feature-name, do all the work and testing in that branch and then when I am happy with it I merge it to develop. Contributors can do the same by branching from develop and submitting a pull request from their branch. When you submit a pull request you are effectively asking me to merge the feature branch into the develop branch.

Typically, if a PR is found to contain a bug after the fact I would not revert it, but instead fix or wait for a fix. But this one created such a serious performance problem that I thought it was best to remove it entirely temporarily until the problem could be solved.

pfried, 2017-03-20:
you are right, a bit unsatisfying that my commits caused that much trouble, i would have liked to fix it quickly, but your were so fast i had not enough time to provide a fix

vonnieda, 2017-03-20:
It's nothing to worry about. Bugs creep in. I'll review the new PR tonight and get it merged.

cri-s, 2017-03-27:
Just generated some errors from unknown xml in machine.xml
Very funny the LOL`s  . I know the error, the point is that there must be something wrong on how the
logger is interfaced to the code. 

Cris

ot have a match in class org.openpnp.machine.reference.ReferenceMachine at line
359)
        at org.openpnp.model.Configuration.load(Configuration.java:268)
        at org.openpnp.gui.MainFrame.<init>(MainFrame.java:508)
        at org.openpnp.Main$1.run(Main.java:94)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionP
rivilege(ProtectionDomain.java:80)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThre
ad.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.
java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThre
ad.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: org.simpleframework.xml.core.ElementException: Element 'signalers' do
es not have a match in class org.openpnp.machine.reference.ReferenceMachine at l
ine 359
        at org.simpleframework.xml.core.Composite.readElement(Composite.java:527
)
        at org.simpleframework.xml.core.Composite.readElements(Composite.java:44
5)
        at org.simpleframework.xml.core.Composite.access$400(Composite.java:59)
        at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:13
83)
        at org.simpleframework.xml.core.Composite.read(Composite.java:201)
        at org.simpleframework.xml.core.Composite.read(Composite.java:148)
        at org.simpleframework.xml.core.Composite.readVariable(Composite.java:62
3)
        at org.simpleframework.xml.core.Composite.readInstance(Composite.java:57
3)
        at org.simpleframework.xml.core.Composite.readUnion(Composite.java:549)
        at org.simpleframework.xml.core.Composite.readElement(Composite.java:532
)
        at org.simpleframework.xml.core.Composite.readElements(Composite.java:44
5)
        at org.simpleframework.xml.core.Composite.access$400(Composite.java:59)
        at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:13
83)
        at org.simpleframework.xml.core.Composite.read(Composite.java:201)
        at org.simpleframework.xml.core.Composite.read(Composite.java:148)
        at org.simpleframework.xml.core.Traverser.read(Traverser.java:92)
        at org.simpleframework.xml.core.Persister.read(Persister.java:625)
        at org.simpleframework.xml.core.Persister.read(Persister.java:606)
        at org.simpleframework.xml.core.Persister.read(Persister.java:584)
        at org.simpleframework.xml.core.Persister.read(Persister.java:543)
        at org.simpleframework.xml.core.Persister.read(Persister.java:521)
        at org.simpleframework.xml.core.Persister.read(Persister.java:426)
        at org.openpnp.model.Configuration.loadMachine(Configuration.java:386)
        at org.openpnp.model.Configuration.load(Configuration.java:261)
        ... 16 more
Exception in thread "Thread-7" java.lang.RuntimeException: java.nio.file.AccessD
eniedException: C:\Users\Gast\AppData\Local\Temp\4400428494890139292\nu\pattern\
opencv\windows\x86_32\opencv_java2411.dllLOGGER ERROR: Failed to write log entry
 (java.io.IOException: Stream Closed)LOGGER ERROR: Failed to write log entry (ja
[... 388 lines cut ...]
)LOGGER ERROR: Failed to write log entry (java.io.IOException: Stream Closed)LOG
GER ERROR: Failed to write log entry (java.io.IOException: Stream Closed)LOGGER
ERROR: Failed to write log entry (java.io.IOException: Stream Closed)LOGGER ERRO
R: Failed to write log entry (java.io.IOException: Stream Closed)
Exception: java.lang.StackOverflowError thrown from the UncaughtExceptionHandler
 in thread "Thread-7"

C:\Users\Gast\Desktop\openpnp-develop>"%JAVA_HOME%\bin\java" -jar target\openpnp
-gui-0.0.1-alpha-SNAPSHOT.jar
Mõr 27, 2017 4:01:48 PM java.util.prefs.WindowsPreferences <init>
WARNUNG: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0
x80000002. Windows RegCreateKeyEx(...) returned error code 5.
Exception in thread "Thread-7" java.lang.RuntimeException: java.nio.file.AccessD
eniedException: C:\Users\Gast\AppData\Local\Temp\4111740394498358713\nu\pattern\
opencv\windows\x86_32\opencv_java2411.dll
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLOLO
LOLOLOLOLOLOLOLOLOLO
Exception: java.lang.StackOverflowError thrown from the UncaughtExceptionHandler
 in thread "Thread-7"

vonnieda, 2017-03-27:
Indeed, at first glance I'd say that the logger probably tries to send output to System.out or System.err when it encounters an error, which causes an infinite loop and stack overflow.

pfried, 2017-03-27:
One of the writer must have an issue with writing to its stream, but this could be the Console logger or the File Logger. Seems that we need to do a check if the stream is writable, or we catch all the errors without an output to the console. I had Exceptions before (the double line issue) but never saw a behaviour like @cri-s

I will check if I can reproduce the behaviour by setting an invalid config

pfried, 2017-03-27:
Cannot reproduce it, @cri-s Could you apply the path below and check if it might solve the problem, only a quickshot

```
Index: src/main/java/org/openpnp/logging/ConsoleWriter.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- src/main/java/org/openpnp/logging/ConsoleWriter.java	(date 1490578123000)
+++ src/main/java/org/openpnp/logging/ConsoleWriter.java	(revision )
@@ -36,7 +36,10 @@
 
     @Override
     public void write(final LogEntry logEntry) {
-        getPrintStream(logEntry.getLevel()).print(logEntry.getRenderedLogEntry());
+        PrintStream stream = getPrintStream(logEntry.getLevel());
+        if(!stream.checkError()) {
+            stream.print(logEntry.getRenderedLogEntry());
+        }
     }
 
     @Override

```

vonnieda, 2017-03-30:
I may have reproduced this, or something quite like it. It only happens on Windows, because of another issue. When the OpenCV library shuts down it tries to delete a temporary file and gets an access denied. This causes an exception to be thrown, which the new logging stuff tries to log. This results (somehow) in a StackOverlowException. I can reproduce this simply by running OpenPnP on Windows and then exiting normally. I get a StackOverflowException in the console. Running under the debugger lets me see that this is happening somewhere in SystemLogger.

[image]

[image]

I think this may be related to #442

vonnieda, 2017-03-31:
That registry error is common, I've been seeing that on Windows for years and it doesn't seem to be a problem. I'm not sure why you would be getting a stack overflow on startup. Have you tried backing up and removing machine.xml?

cri-s, 2017-03-31:
yes deleted config files, recompiled, nothing, then deleted the branch
in registry and it worked again.
I can reproduce it, doing a pipeline in standalone piepeline editor,
and then openpnp don't starts giving out several thousand logging
errors about registry access before doing stack overflow.

vonnieda, 2017-03-31:
Okay, so it must be that the standalone editor is saving a value in the preferences (for the current directory) that somehow kills OpenPnP's reading of preferences. Can you please file this as a new issue?

vonnieda, 2017-04-02:
I have temporarily removed system logging as it was causing a whole host of issues. See comments in https://github.com/openpnp/openpnp/commit/be475a25171c61d99cd6ec97899a7e2ea303304c

pfried, 2017-04-17:
I want to provide a fix for it and also for the StackOverflow. I am pretty busy right now (also due to building my own PnP machine 🤓, will do a forum post on this later on). Expect the fixes within the next 2-3 weeks

vonnieda, 2017-06-14:
Newline issue fixed in #579

vonnieda, 2017-06-14:
I'm going to go ahead and close this as completed. All the major functionality has been added and there have no further requests for functionality. Future requests can be handled on a one by one basis.

closed 2017-06-14 by vonnieda