openpnp/openpnp #289: Feature request: Socket allowing start board mounting cycle

la3pna, 2016-06-13:
As discussed on IRC, I'd like the idea of a simple socket that could be used to get openPNP to run a board with a given command. 
HTTP / REST route would be fine: curl -XPOST http://localhost:8080/api/job/start
Preferably with a way to report when the board are finished mounting the parts.

This should allow for remote run of board, and even complete automated systems.

vonnieda, 2016-06-13:
This can be added quite easily with the experimental web API code in the web branch: https://github.com/openpnp/openpnp/tree/feature/web. That code was started to provide a complete REST interface to the program, but adding just this along with some basic informational routes would be a great start to getting the web interface going.

tritao, 2016-06-15:
Nice idea.

If the goal is to provide a comprehensive API over time, it might be interesting to consider something like [gRPC ](http://www.grpc.io/), it supports Java and is very easy to consume from all popular languages.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda