openpnp/openpnp #295: no defined transitions from cleanup for initialize  after Job start inside PnPJobProcessor

cri-s, 2016-07-03:
After starting job, the error 
no defined Transistion from cleanup for initialize 
comes out, new installation, added driver (grbl) and other small adjustments.

cri-s, 2016-07-03:
[image]

cri-s, 2016-07-03:
[machine.zip](https://github.com/openpnp/openpnp/files/345318/machine.zip)

I supposed machine.txt was uploaded, but it don't seem so, so i have zipped it.

vonnieda, 2016-07-04:
This machine.xml contains many changes to the codebase. soft limits, feed rate, actuator on camera, changer location. I don't have the code for all of these changes so I cannot test this. Please test on a clean codebase and see if you still get the error.

cri-s, 2016-07-04:
yes, same on clean codebase.
it happens if stopping job, reduce enabled parts from job, and restarting it.
The first time it run it works. Do you have some idea about the reason
of this issues ?

cri-s, 2016-07-04:
To the modifications, custom grbl driver, have tested gcode driver but
that was not able
to talk to grbl. Actuator on camera, if using camera, it must switch
off lights and switch on
uplights. For that are the actors on camera. Dragfeeder was modified
to retract the drag needle
a bit in order to have less probability to jam and added additional
timing using optional setting
with dummy Actuator. ReferenceCamera had some errors with mat handling
and crop/offset handling, nozzle changer needed 2 additional moves in
order to not crash on nozzle when entering or exit nozzle changing in
order it have clean travel to nozzle holder and drag pin don't hit the
nozzles.

2016-07-04 10:01 GMT, Cri S <phone.cri@gmail.com>:

vonnieda, 2016-07-04:
@cri-s What do you mean by reduce enabled parts? What are the exact steps required to reproduce the issue? Please see https://github.com/openpnp/openpnp/blob/develop/CONTRIBUTING.md#issues-bug-reports-and-feature-requests for help.

cri-s, 2016-07-05:
I mean modify type from place to ignore. Actually its testing, so
different things are not setup correctly, only remote access, not the
best situation to make minimal test case.
I cannot compile it here unmodified because of opencv bug, and
modification is heavy.
Now I have checked fsm code and think to have found the issue. It seem
exception handling is not consequent implemented and give some side
effects. It seems. Knowing that I can thereof build the minimal error
case, at least one of several I have seen.
On the other side, on production setup this error is unlike to be
present and other errors,
like core dump of pipeline queue is more frequent, so maybe it's not
worth investigation.
If the fsm is reset on loading the job, i think this should be ok.

cri-s, 2016-07-20:
Problem related to unexpected exceptions from nozzletip.calibration. if setup correctly,problem disappear.

closed 2016-07-20 by cri-s

sp-apertus, 2017-01-15:
I just want to report that this issue is still present with the latest dev version and pops up after pressing the job "run" button. I cannot say exactly what my steps were to trigger the issue but previous reports by cri-s: "it happens if stopping job, setting some components to "ingore" from job, and restarting it." sounds very much like something I could have done:
 
[image]

vonnieda, 2017-01-15:
@sp-apertus This entire system is going to be ripped out and replaced very soon. It was a mistake to begin but got us past a hurdle. Major rework is being done in #280. I'm going to leave this closed unless someone can provide instructions to reproduce since I don't think it's worth the time tracking it down if it's going to be replaced soon anyway.

sp-apertus, 2017-01-15:
Right, no problem.