openpnp/openpnp #298: CvPipline Stages need better protection against null results

vonnieda, 2016-07-07:
See discussion at: https://groups.google.com/d/msgid/openpnp/0c66d166-a9c6-4eb6-801c-902cd425c6c0%40googlegroups.com

Some of the stages that call sensitive OpenCV methods need better protection to avoid crashing the program. MaskHsv is the one referenced in the discussion. 

Need to investigate what part of MaskHsv is causing the crash and then protect for that a similar errors. Should be able to test by feeding a black image into MaskHsv.

cri-s, 2016-07-08:
I don't think checking args is the key to protect to this error.
OpenCV is really sensitive to this type of errors.
Example:
Using Gaussian Blur image with even argument, as example 12 or 4 or whatever.
The result is a core dump.
Similar, if depth of image, size of image is not a multiple of 8, ... is not correct for that operation,
then core dump is the result.
I think, the only protection is where image is required as example imageRead or imageCapture
and null is returned, and for pipeline editor, it should fork external java process that do the required action in order that if it crashes, nothing is lost.

vonnieda, 2016-07-08:
@cri-s I agree, it's not perfect, but calling OpenCV externally is not an option. That will add too much complexity and delay. We can provide checks against the most common errors and since this is only an issue when the pipeline is being modified (not when it's just running) it should mitigate most of the problems.

cri-s, 2016-07-08:
My suggestion was to call only the pipeline editor externally, not during
work, and changing a bit the dependency it works really good, at least on
Linux.

vonnieda, 2016-07-08:
@cri-s Ah, I see. Yes, that is a viable solution. Can you explain what you mean by "changing a bit the dependency"?

cri-s, 2016-07-09:
Can you help me to unlook simpleblob to look only for black circles ?
How do you generate this xml using simpleframework ?
<?xml version="1.0"?>
<opencv_storage>
<thresholdStep>10.</thresholdStep>
<minThreshold>50.</minThreshold>
<maxThreshold>220.</maxThreshold>
<minRepeatability>2</minRepeatability>
<minDistBetweenBlobs>10.</minDistBetweenBlobs>
<filterByColor>1</filterByColor>
<blobColor>0</blobColor>
<filterByArea>1</filterByArea>
<minArea>25.</minArea>
<maxArea>5000.</maxArea>
<filterByCircularity>0</filterByCircularity>
<minCircularity>8.0000001192092896e-01</minCircularity>
<maxCircularity>3.4028234663852886e+38</maxCircularity>
<filterByInertia>1</filterByInertia>
<minInertiaRatio>1.0000000149011612e-01</minInertiaRatio>
<maxInertiaRatio>3.4028234663852886e+38</maxInertiaRatio>
<filterByConvexity>1</filterByConvexity>
<minConvexity>9.4999998807907104e-01</minConvexity>
<maxConvexity>3.4028234663852886e+38</maxConvexity>
</opencv_storage>

cri-s, 2016-07-09:
maybe this works.
Requirement is , this as separate file.
<?xml version="1.0"?>
<opencv_storage>
<thresholdStep>10.</thresholdStep>
<minThreshold>50.</minThreshold>
<maxThreshold>220.</maxThreshold>
<minRepeatability>2</minRepeatability>
<minDistBetweenBlobs>10.</minDistBetweenBlobs>
<filterByColor>1</filterByColor>
<blobColor>0</blobColor>
<filterByArea>1</filterByArea>
<minArea>25.</minArea>
<maxArea>5000.</maxArea>
<filterByCircularity>0</filterByCircularity>
<minCircularity>8.0000001192092896e-01</minCircularity>
<maxCircularity>3.4028234663852886e+38</maxCircularity>
<filterByInertia>1</filterByInertia>
<minInertiaRatio>1.0000000149011612e-01</minInertiaRatio>
<maxInertiaRatio>3.4028234663852886e+38</maxInertiaRatio>
<filterByConvexity>1</filterByConvexity>
<minConvexity>9.4999998807907104e-01</minConvexity>
<maxConvexity>3.4028234663852886e+38</maxConvexity>
</opencv_storage>

cri-s, 2016-07-09:
can you help to unlook simpleblob code ?
How to generate this separate file with simpleframework.xml
http://pastebin.com/wyAJSndf

vonnieda, 2016-07-09:
@cri-s Please post this question to a new issue or to the mailing list. It is not related to this issue.

vonnieda, 2017-04-14:
Recently changed CvPipeline to return a default image in the case of a null or zero size image, which fixes this for all the most common cases.

closed 2017-04-14 by vonnieda