openpnp/openpnp #299: feeder name mismatch

cri-s, 2016-07-10:
when writing script for programming parts, i discovered one problem.
It's not really a problem, but when using scripts, it can be.
image: https://s32.postimg.org/avtkhydp1/2016_07_11_005542_1280x800_scrot.png

I have made 4 feeder, pgm_chip , pgm_ok , pgm_fail, pgm_dev .
If i search that feeder using machine.getFeeder("pgm_chip"), it don't find nothing.
Listing all feeders on script using machine.getFeeders() it get the follow:
// other feeders cut out
feeder10: org.openpnp.machine.reference.feeder.ReferenceTubeFeeder@b9205c
feeder11: org.openpnp.machine.reference.feeder.ReferenceTubeFeeder@1c9c0aa
feeder12: org.openpnp.machine.reference.feeder.ReferenceTubeFeeder@1f3be23
feeder13: org.openpnp.machine.reference.feeder.ReferenceTubeFeeder@107e0d

ps: how to enable the log tab under linux ? it is currently supposed to work, under unix nothing
is outputted and it's not possible to change log level.

vonnieda, 2016-07-11:
getFeeder takes a feeder ID, not a name. See http://openpnp.github.io/openpnp/develop/org/openpnp/spi/Machine.html#getFeeder-java.lang.String-

You will have to iterate through the list of feeders and check the names to find the one you want. I'll leave this ticket open as a request to add a getFeederByName() like we have for Actuators: http://openpnp.github.io/openpnp/develop/org/openpnp/spi/Machine.html#getActuatorByName-java.lang.String-

tracklogic, 2016-08-02:
Implemented on branch, will initiate a pull request to get into main develop shortly.

vonnieda, 2016-08-04:
Fixed in #313

closed 2016-08-04 by vonnieda