openpnp/openpnp #301: Test Suite may be using incorrect configuration directory on Windows

vonnieda, 2016-07-11:
See discussion at https://groups.google.com/d/msgid/openpnp/CAMLuGALdWa6hL_LXUotsw8f_T4aek%2B%3DezQnc9Lw0VGxmq3dLDw%40mail.gmail.com?utm_medium=email&utm_source=footer

It looks like the test suite is failing to use a new, empty temporary directory on Windows. Need to do some Windows specific testing to verify. To verify, add bad data to the user's machine.xml and see if the tests will run. They should use their own built in configs and not the user's global one.

PeeJay, 2016-07-19:
I use IDEA on win64 to do my openpnp debugging, and I always get a similar error on exit (to the ones in the link) when the opencv module tries to clean up, no permission to delete. 
I've been trying to get ffmpeg support working with opencv, which requires copying an additional dll along with the java library, but it doesn't get loaded from the temp directory (it copies ok though). Originally I thought it was a windows dynamic linking problem, but it sounds exactly like this bug report. I'll keep investigating and see if I find anything

vonnieda, 2016-07-19:
@PeeJay There are two errors in the log. The permission one comes from OpenCV and is a known bug that doesn't cause any problems. It just fails to delete a temporary file.

For your ffmpeg issue, you'll need to actually rebuild OpenCV with support turned on or it won't know to look for the DLL. If you'd like to discuss that further, please create a new issue or thread on the mailing list.

vonnieda, 2017-04-14:
Verified that this is working as intended.

closed 2017-04-14 by vonnieda