openpnp/openpnp #302: Don't crash on incorrect Machine.xml

PeeJay, 2016-07-19:
It would be great if Openpnp didn't crash if there are unexpected or missing settings found in Machine.xml eg.
`java.lang.Exception: Error while reading machine.xml (Attribute 'use-external-open-CV' does not have a match in class org.openpnp.machine.reference.ReferenceMachine at line 2)'
A suggestion would be to alert the user about the problem and ask if the setting should be reset to default, or quit the program.

vonnieda, 2016-07-19:
@PeeJay Can you send me the machine.xml that crashes the program? I've had support for graceful failure with user notification in for a long time and I have not seen an XML related crash since I added it.

PeeJay, 2016-07-27:
Adding any attributes or elements seems to do it.

For example:
'<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine" speed="1.0">
   <goofy>
   
   </goofy>
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="22964dce-252a-453e-8106-65db104a0763" name="H1">
            <nozzles>'

Results in:
`WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
java.lang.Exception: Error while reading machine.xml (Element 'goofy' does not have a match in class org.openpnp.machine.reference.ReferenceMachine at line 3)
	at org.openpnp.model.Configuration.load(Configuration.java:259)
	at org.openpnp.gui.MainFrame.<init>(MainFrame.java:493)
	at org.openpnp.Main$1.run(Main.java:96)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: org.simpleframework.xml.core.ElementException: Element 'goofy' does not have a match in class org.openpnp.machine.reference.ReferenceMachine at line 3
	at org.simpleframework.xml.core.Composite.readElement(Composite.java:527)
	at org.simpleframework.xml.core.Composite.readElements(Composite.java:445)
	at org.simpleframework.xml.core.Composite.access$400(Composite.java:59)
	at org.simpleframework.xml.core.Composite$Builder.read(Composite.java:1383)`

vonnieda, 2016-07-27:
Does this actually crash the program? Or do you get an error dialog that
explains how to reset the configuration?

Jason

PeeJay, 2016-07-27:
Oh I see what you mean now. Yes, I get the dialog with instructions on how to delete the config.
That just doesn't seem very helpful though. If openpnp is upgraded (or during development) we have to manually edit machine.xml if we don't want to loose our existing config.

vonnieda, 2016-07-27:
PeeJay, in general any forward upgrades are backwards compatible. In that I
mean that upgrading to a newer version should never require you to modify
the config. I agree, though, that the dialog can certainly be more useful.
I'll update this ticket to indicate that. I just wanted to address the
concern that there was a possible crash.

Jason