openpnp/openpnp #306: Automatically calibrate "Units per Pixel" on cameras

mrandt, 2016-07-28:
Units per pixel could be automatically calibrated as follows:

For head camera:

1. Jog machine above an easily detectable optical mark, a fiducial would work great
2. Use vision to detect exact location of mark(like in fiducial detection)
3. Diagonally move machine in XY direction by a known offset 
4. Detect location of mark again and calculate distance relative to location in step 2.
5. Calculate units-per-pixel for camera

For bottom / table camera:

1. Jog nozzle tip above camera and lower to PCB surface level
2. Use vision to detect center of nozzle tip
3. Diagonally move machine in XY direction by a known offset 
4. Detect location of nozzle tip again and calculate distance relative to its location in step 2
5. Calculate units-per-pixel for camera

The steps 4. and 5. could be repeated multiple times with different distances and directions to achieve even better accuracy, especially if camera lense has strong distortion.

issus, 2016-08-06:
I don't know that this will give accurate results, as you'll be dealing with the parallax as well as lens distortion (especially with lenses that tend towards fisheye). You'd be better off finding a fiducial on the board of known size and using that, as the copper should be etched with a micron or so of spec - depending on the board fab.

If you were to do this by moving the camera/nozzle around, you'd have to do it in very small movement so as to stay relatively centred in the screen.

mrandt, 2016-08-08:
I understand your concerns. Parallax and lense distortion are an issue in any operation which uses camera measurement; bottom vision alignment of parts, feeders, etc. - so if we assume camera is accurate enough to do these measurements; it should be safe to assume that the same camera can be used for px-mm conversion. Also, the error you mentioned applies as well to the "ruler" or "graph paper" method most users utilize; doesn't it?

I like the idea of using the fiducial area; on the other hand it is a relatively small sample which may cause problems with lower resolution cameras.

By the way, LitePlacer RMOD fork supports px-mm calibration for both up and down camera using the method I described and (at least on my machine) the results worked very well; much better than any manual measurement I had done before.

Last not least, it is a question of better usability which could help new users to get their machine running much quicker.

markmaker, 2022-02-03:
Implemented by #1297 and #1335.

closed 2022-02-03 by markmaker