openpnp/openpnp #307: Semi-automatic measurement of nozzle-tip to camera offset

mrandt, 2016-07-28:
I currently measure the offset as follows:

1. Put a piece of carbon paper (write-through type) on table surface
2. Jog the head above paper and lower the nozzle, so it will leave a circular mark on table
3. Remove the carbon paper, note machine coordinates
4. Jog camera above the mark until its centered
5. Note new machine coordinates and calculate offset in X and Y direction

I think this process could be supported by a small wizard; mark detection and calculation (steps 4. and 5.) could probably be automated.

In step 4, the user could jog the camera near the mark, using vision system camera could be centered over it exactly.

vonnieda, 2016-08-27:
Coming up with a good set of documentation and a standard method for doing this would be helpful, as well. The wizard doesn't have to be super configurable, it just needs to work for some case that is easily accessible. For instance, we can tell users to use a piece of putty or bubble gum since these can be found all over the world.

pfried, 2018-08-19:
I started working on this "feature". I plan to put it on a tab under the machine config -> nozzle

A small set of instructions as above and a small guided wizard.