openpnp/openpnp #308: Configure and enforce "safe" and "no-go" areas

mrandt, 2016-07-29:
The software could have an option to configure and enforce a safe area for head movement (X, Y and Z).

Functionality could include:
- Setting min / max X coordinates for machine
- Setting min / max Y coordinates for machine
- Setting min / max Z-level (per nozzle or even per nozzle tip?)
- Optional / Extension: Defining additional "no-go" areas by providing X, Y, width, length, and height; e.g. to protect feeders or other features on table
- If option is enabled, software should refuse jogging unless it has been properly homed (thus coordinate system is assumed reliable) - whenever the connection to controller is cut or controller is reset, homing should be enforced again
- Software should refuse any command (either jogging, cam point + click, placement job) that would take the head outside the safe areaor into any of the no-go-areas and it should especially prevent users from jogging (and crashing) Z-axis into the table surface or bottom (table) camera...

It seems users (including myself) should be protected from themselves :-(

BTW: I think for the Z-axis 100mm step does not make sense either; should probably be blocked.

tracklogic, 2016-08-03:
I think some of this may have to be implemented in the motion controller firmware.. OpenPNP will say goto X,Y but it has no control if a no-go area is in the middle of the path the motion controller decides to take?

vonnieda, 2016-08-03:
Indeed. I think we can probably do soft limits pretty easily (and we once
did, years ago) and we can probably do no go destinations but we can't
really control the path the motion controller takes. So, for instance, we
could say things like "Don't allow commands to target the inner bounds of
this rectangle" but if the motion controller happens to go through such a
rectangle on the way to a legal destination that's out of our control. And
I think that's okay. Just having the basics would be a big improvement.

For anyone wanting to tackle this, the code will probably land in
ReferenceHead and may require some rework of the offsets system with regard
to drivers.

cri-s, 2016-08-04:
Usually the nozzle changer station require deeper traveling as board limit.
It's easy implementing it inside drivers and many fw have softlimit already implemented.

mrandt, 2016-08-08:
OK, I get the difference between "soft limits" (do not forward coordinates outside safe area) and "no go zones" (do not go to or traverse a certain area).

I also appreciate that the first would be much easier than the latter - and I also think the first is more important.

I have crashed my Z-axsis into the table surface multiple times already and manged to "bang" the carriage into one of the slides... 

I consider it a matter of usability to NOT let users do stupid things in software - and most CNC I worked with do have soft limits to prevent crashes.

cri-s, 2016-08-08:
On the Grbl driver i have:
Type 1 is normal softlimit
Type 2 is it goes out of safe Zone.
If just using normal softlimits, then softlimits1 need to have the
same values as
softlimit0. SoftLimit0 must have contains all areas, softlimitN need
to delimit safe area.
It can be extended to more areas. The important point is, it's not
allowed to travel from one
area to annother, the whole move must be done inside one area only.

  @Element(required=false)
    private Location softLimitMin0  = new Location(LengthUnit.Millimeters);
    @Element(required=false)
    private Location softLimitMin1  = new Location(LengthUnit.Millimeters);
    @Element(required=false)
    private Location softLimitMin2  = new Location(LengthUnit.Millimeters);
    @Element(required=false)
    private Location softLimitMax0  = new Location(LengthUnit.Millimeters);
    @Element(required=false)
    private Location softLimitMax1  = new Location(LengthUnit.Millimeters);
    @Element(required=false)
    private Location softLimitMax2  = new Location(LengthUnit.Millimeters);
    @Attribute(required=false)
    private boolean softLimitsEnabled=false;

     if (softLimitsEnabled)
                if(checkSoftLimit(x,y,z,c)!=0) {
                        throw new Exception(
                                String.format("Softlimit error type
%d, X=%3.3f Y=%3.3f Z=%3.3f
C=%3.3f",checkSoftLimit(x,y,z,c),x,y,z,c));
                }

      private int checkSoftLimit(double x0, double y0, double z0,
double c0) throws Exception {
        if (
                                x0 < softLimitMin0.getX() || x0 >
softLimitMax0.getX() ||
                                y0 < softLimitMin0.getY() || y0 >
softLimitMax0.getY() ||
                                z0 < softLimitMin0.getZ() || z0 >
softLimitMax0.getZ() )
                        return 1;
                        if(
                                x0 >= softLimitMin1.getX() && x0 <=
softLimitMax1.getX() &&
                                y0 >= softLimitMin1.getY() && y0 <=
softLimitMax1.getY() &&
                                z0 >= softLimitMin1.getZ() && z0 <=
softLimitMax1.getZ() &&
                                x  >= softLimitMin1.getX() && x  <=
softLimitMax1.getX() &&
                                y  >= softLimitMin1.getY() && y  <=
softLimitMax1.getY() &&
                                z  >= softLimitMin1.getZ() && z  <=
softLimitMax1.getZ() ||

                                x0 >= softLimitMin2.getX() && x0 <=
softLimitMax2.getX() &&
                                y0 >= softLimitMin2.getY() && y0 <=
softLimitMax2.getY() &&
                                z0 >= softLimitMin2.getZ() && z0 <=
softLimitMax2.getZ() &&
                                x  >= softLimitMin2.getX() && x  <=
softLimitMax2.getX() &&
                                y  >= softLimitMin2.getY() && y  <=
softLimitMax2.getY() &&
                                z  >= softLimitMin2.getZ() && z  <=
softLimitMax2.getZ() )
                        ; else
                        return 2;
                        return 0;
        }

vonnieda, 2016-08-27:
A note to anyone who works on this: The code should be put into Head, Nozzle, Actuator, Camera, etc. instead of in the driver. Or possibly it should be in a common driver base class. We don't want to add it specifically to a single driver because then it is driver specific and needs to be added to all of them.

cri-s, 2016-08-27:
It is more appropriate to rename moveTo to move inside Drivers and
implement moveTo
inside ReferenceDriver.java that then calls move . I have it done times
before and it gives some advantage.

vonnieda, 2016-08-27:
Do you mean applying head offsets in ReferenceDriver? If so, I agree. I
think putting the head offset application in the driver was a mistake and I
want to move it out.

Jason

cri-s, 2016-08-27:
I have wanted abstraction of moveTo too.
For that I had renamed all moveTo  to move inside drivers
and only on ReferenceDriver the move to calling move.
That's allow having camwheel mapping,
Calling dwell if head is camera or actor if move is on Z , otherwise dwell
is called only inside pick and place for moves.
Other thing I have added, if automatic placing is running and moving is on
nozzle, on moving inside PCB space, configured separately, move camera over
it, take photo, save position and  move to .nozzle . this if xy move.
On moving up, after doing that move go over saved position and take another
snapshot.
Save null into saved location.
This allows optional checking placing accuracy and automaticall check if
part was lost. If reference picture is same as target (psrn comparison)
then part was lost.
Limits including height limits and other small additions, this to all
driver.
Do you mean applying head offsets in ReferenceDriver? If so, I agree. I
think putting the head offset application in the driver was a mistake and I
want to move it out.

Jason

> It is more appropriate to rename moveTo to move inside Drivers and
> implement moveTo
> inside ReferenceDriver.java that then calls move . I have it done times
> before and it gives some advantage.
>
> > A note to anyone who works on this: The code should be put into Head,
> > Nozzle, Actuator, Camera, etc. instead of in the driver. Or possibly it
> > should be in a common driver base class. We don't want to add it
> > specifically to a single driver because then it is driver specific and
> > needs to be added to all of them.
> >
> > <
UFpZex82k7X2IlObfks5qkGlrgaJpZM4JYdhE
ABIKc2cGFGYX28OtCq1D5Lgb532R26PGks5qkHiIgaJpZM4JYdhE>

—
<https://github.com/openpnp/openpnp/issues/308#issuecomment-242932219>, or mute
the thread

machinekoder, 2017-09-04:
@mrandt https://github.com/openpnp/openpnp/pull/598 implements a feature to prevent the nozzle from moving into any boards. You could create "danger zone" boards with the correct Z height to achieve what you want,

jetty840, 2018-12-09:
I'm thinking this is really 2 issues, not 1: Soft Machine Axis Limits (which is a constraint of the hardware) and No Go Zones.  

I.E. Machine X/Y/Z/C limits should be in a common driver class  just before the driver moves the hardware (i.e. after squareness correction, transformation etc), e.g. in the gcode driver it would be before sendGCode but it would need to be implemented generically for all drivers.

and 

No Go Zones should be at a higher level (i.e. Head etc.).  No Go Zones would also need names and some kind of override for situations where entering a no go zone entry is required (e.g. during a nozzle change).

Also it would be possible to figure out if a trajectory between 2 points intersects with the 3D volume of a no go zone with this algorithm, so the situation where a move goes through the no go zone and out the other side could be prevented assuming the no go zone is cuboid with this technique: https://en.wikipedia.org/wiki/Line–plane_intersection  as essentially there are 6 planes (the 6 sides of the cuboid) and intersection of the trajectory with any of those planes, the 3D derivative of this is used in raytracing...  a ray intersecting an Axis Aligned Bounding Box (ray-AABB):  https://www.scratchapixel.com/lessons/3d-basic-rendering/minimal-ray-tracer-rendering-simple-shapes/ray-box-intersection

All this fails if the user manually starts issuing GCode commands via the command line, as openpnp isn't aware of the controllers position, only what it's sent to the controller, but we could probably live with that.

vonnieda, 2018-12-09:
@jetty840 Agreed - it's two issues. Soft limits is pretty easy, and honestly I think "no go zones" is a neat idea but not really that important. 

The problem is that currently head offsets get applied at the driver level, and this is a mistake. It shouldn't have been that way from the start, but I think I was considering some kind of Transformers style morphing machine when I designed it :) 

Head offsets need to move up to the head, and then this will be trivial to implement.

jetty840, 2018-12-10:
Also thinking backlash should be cleaned up too.  Currently it's in the gcode driver and should be in all them.  Also rather than have a Backlash Regex with a 2 line gcode, have the driver just do 2 moves when the compensation != 0.

vonnieda, 2019-02-13:
X and Y soft limits have now been added: https://github.com/openpnp/openpnp/commit/3da71fbc71769196143e1f0297ee4cd941848faa

vonnieda, 2019-05-27:
With soft limits now added I'm closing this one. Adding no-go areas would require a pretty major redesign of OpenPnP and would be controller dependent, so I don't think it will be implemented. If anyone feels very strongly that soft limits are not good enough, please reopen and comment.

closed 2019-05-27 by vonnieda

noahbliss, 2025-03-30:
Hey y'all, not sure if this is solved yet, but for the sake of nozzle change speed and space-limit problems, my nozzle rack is often in the "shortest path" for moves. Because of the limited z movement of my particular machine, the nozzle rack _must_ be an unsafe height, obviously without the "keep out" feature this results in a ton of crashes. 

Pinging @vonnieda for visibility. If this should be a new issue, let me know.

noahbliss, 2025-03-30:
Potentially related: 

There are some feeders on both ends of my machine that can only be accessed by either N1 or N2 but not both. It's also impossible to center the camera over these feeders. Is there any way to keep the head from crashing and to potentially "whitelist" which nozzle can interact with a specific feeder? Getting CV correction/info for these without crashing would also be awesome (by bringing it into view without being centered in the camera)

noahbliss, 2025-03-31:
As a temporary solution for the nozzle rack issue, tossed this together. It's a gcode proxy that accepts a rectangle. It reroutes movements that would collide with it, but accepts movements that are intended for it. 

https://github.com/noahbliss/antisocial-gcode