openpnp/openpnp #317: Automated Optical Inspection

issus, 2016-08-06:
I'd like to see automated optical inspection of the placement job, and even the ability to put the reflowed board in again for post reflow inspection.

Perhaps the camera could stitch a full image of the board prior to placement and then diff this to an image of the board after placement - or take a picture of each placement location at the start of the job, and then at the end of the job. This could be as simple as presence detection, or could recognise that the part is centred and rotated correctly.

I don't think the machine should try to fix any issues, simply alert the user.

mrandt, 2016-08-08:
Maybe even start simpler - detect failed pick and / or placement using camera. After picking up the part from the feeder (tape strip, whatever), use camera to check that the part is actually gone - and assume it's at the nozzle.

Same for placement: Take a snapshot before and after placement; should be very easy to detect that the pads are now partially covered.

I am aware that this will take much more time and drastically reduce CPH, but at the same time it may well reduce errors and ruined boards. 

Many times I don't care how long a placement job actually takes, as long as I don't have to babysit the machine and can do other stuff at the same time. In those situations low error rate is more important than time.

issus, 2016-08-08:
Failed pick is another open issue already. Perhaps the vision after the placement detecting no part would fall into this category?

I see where checking the part after the placement could be beneficial, however I think checking at the end might be better, as it will detect parts that might have been bumped or knocked. For example, a large capacitor gets tapped by a large microcontroller on it's way to being placed. The capacitor has already been placed, so it wouldn't show up as an error - the micro would get placed incorrectly and then detected as a bad placement.

Agreed on not caring about job run time, as long as its done right for critical parts.

tracklogic, 2016-09-07:
@vonnieda - do you have any code already for building a stitched together image of the machine bed - i guess kind of like what the image camera presents already (the sample.png). Be quite useful for setting the machine co-ordinates up offline i think and just general playing about?

vonnieda, 2016-09-07:
Unfortunately I don't. This is something I want to build in to the new UI
eventually so that you can get a view of the machine without a high, wide
angle camera. I know OpenCV has some functionality to do this but I haven't
looked into it:
http://docs.opencv.org/2.4/modules/stitching/doc/stitching.html

Jason

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

estechnical, 2019-03-08:
Hi - cool project! :)

I would be very interested in helping out with this, if any progress has been made that would help in this area since this ticket was closed, it would be nice to discuss this possibility. I have a test setup that's suitable and intend to have a look. Should I open a new ticket or should we discuss continuing here?

vonnieda, 2019-03-10:
Hi @estechnical, I'll reopen this ticket and we can discuss here. I'm interested to hear your thoughts on how this might work.

estechnical, 2019-03-11:
:) thanks!

I've got my test platform set up enough to move and recognise a fiducial, so that's a start towards my testing.

Initially I was going to see what I could get working just by scripting. I hope I can get access to the parts list of the loaded job, motion control and vision from scripting. That would be a good start.

In future, I would like to be able to configure a job/board that specifies something like "already placed" and "inspect" as true, and a simple operation to check for part presence.

Better vision checking possibilities could be added later, such as alignment checks etc.

Vision pipelines for each type of inspection operation would be a great feature to be able to edit everything in the GUI just like the other vision jobs.

Since I'm pretty new to exploring openpnp, please feel free to point out anything that may help me on the way to this.

As other people have previously suggested, checking after reflow would be advantageous as well.

vonnieda, 2019-03-12:
@estechnical You should be able to do the whole thing with scripting, even including pipelines if you like. The scripting system basically has access to every object in the program. I think that's a great way to go for a proof of concept.

estechnical, 2019-03-13:
Superb, I'll see where I get - thanks for the info!

estechnical, 2019-03-15:
I found some time to try scripting in JS and got as far as being able to visit all placement locations on all boards in the currently loaded job and save an image. That feels like a start ;)

I'd like to make a simple comparison/find with vision eg find part footprint template in the image. Is there anywhere that something similar happens in existing code? I'm reading through the code but I'm sure I've barely scratched the surface yet!

Script so far: https://gist.github.com/estechnical/1eb549c64543f84169b3d12f80027750
:)

estechnical, 2019-03-15:
Would anyone mind helping me pull together some script examples for vision tasks?

vonnieda, 2019-03-16:
Hi @estechnical, I'd recommend asking on https://groups.google.com/forum/#!forum/openpnp rather than here. The discussion group is quite active and has many more people monitoring it than the issues list.

markmaker, 2023-04-08:
See also here:
https://groups.google.com/g/openpnp/c/P4tUFLu7qFw/m/DfDBg-UzAwAJ