openpnp/openpnp #318: Rudimentary 'anti-backlash' support

redvers, 2016-08-07:
As mentioned on the mailing list, the idea that for those of us with, ahem... less than solid hardware can get better accuracy if the head always approaches from the same direction.

issus, 2016-08-08:
This is also applicable to nozzle rotation, especially for people with geared stepper motors, or not so great stepper drivers.

tracklogic, 2016-08-09:
Is this not a firmware thing? Especially the nozzle rotation... we will rotate the nozzle say to 90* then if we want to go to 45* its up to the firmware if to rotate ccw 45* or if to do a cw 315* to satisfy the G0 R45 command?

vonnieda, 2016-08-27:
I generally think it would be better if this was handled in firmware, but since most firmwares do not support backlash compensation, it would be easy enough to add it into OpenPnP. We can either do the "approach from one direction" way, or add in configurable backlash amounts and just correct for that amount when changing direction. This is how Mach 3 does it.

sp-apertus, 2017-01-03:
I took care of this matter and seem to get very good accuracy improvements on my machine (liteplacer).

Pull request: https://github.com/openpnp/openpnp/pull/400

vonnieda, 2017-01-03:
Fixed by #400, needs to be documented in [[GcodeDriver]].

vonnieda, 2017-01-04:
Documented at https://github.com/openpnp/openpnp/wiki/GcodeDriver#backlash-compensation, closing issue.

closed 2017-01-04 by vonnieda

drivelog, 2017-02-12:
How can I set the backlashOffsetX and  backlashOffsetY?

(

2017-02-12 23:55:35 [pool-5-thread-1] org.openpnp.machine.reference.driver.GcodeDriver DEBUG: sendCommand(G0 {BacklashOffsetX:X%.4f} {BacklashOffsetY:Y%.4f} F24750, 7000) => [ok]
2017-02-12 23:55:35 [pool-5-thread-1] org.openpnp.machine.reference.driver.GcodeDriver DEBUG: sendCommand(G0   F24750, 7000)...
2017-02-12 23:55:35 [pool-5-thread-1] org.openpnp.machine.reference.driver.GcodeDriver DEBUG: sendCommand(G0   F24750, 7000) => [ok]

)

vonnieda, 2017-02-12:
@drivelog See https://github.com/openpnp/openpnp/wiki/GcodeDriver#backlash-compensation, and make sure you are running the most recent version of the software. This feature was added only about a month ago, so if you have an older version it will not work.

drivelog, 2017-02-12:
ok i will do so...
Thank you for fast answer

drivelog, 2017-02-12:
OK, Now the Newest Version - MUCH BETTER with Backlash Comp. 
Thank You for ALL!!!

vonnieda, 2017-02-12:
@drivelog Glad to hear it!