openpnp/openpnp #319: Add option to check local fiducials on a board prior to placement

issus, 2016-08-08:
Add a column to the Job -> Boards -> Pick and Place table for checking fiducials before placing the part. If the column is checked, then the down camera will find the two (configurable?) closest fiducials to the part and adjust alignment before placing the part.

This will be useful for high pin count parts, oversize parts, or very fine pitch parts.

jjpickett, 2016-08-11:
After some initial research on this, I believe the best approach is to break this down into smaller proportions.

**Step 1:** Make the system have the option to do a fiducial check before placing a part on the board.
**Step 2:** Adjust the system so the fiducial check is only on the individual board before placing a part on it.
**Step 3:** Adjust the system so the fiducial check is on the closest fiducials to the part before placing the part on the board.

This should allow for a smoother transition and allow for any bugs to be managed and fixed much more efficiently.

vonnieda, 2016-08-21:
Step 1 and Step 2 completed by #327.

vonnieda, 2019-05-27:
Note that the work done in #327 really just added a way to re-check fids on the entire board. It didn't implement local fid checking. That code has now been removed and this feature request is still open.