openpnp/openpnp #321: Tray Feeder Count Not Updating

jjpickett, 2016-08-10:
**Summary:** The Tray Feeders feed count is not being updated when performing a feed and pick.

**Steps:**
1. Go to the bottom panel and select the tab Feeders
2. Create and set up a new Tray Feeder
3. Give the Tray Count some numbers (I used 1 and 3)
4. Select the "Perform a feed and pick on the selected feeder" button.

**Expected Result:** The Feed Count will update its number on the screen.

**Actual Result:** Notice that the Feed Count will display the same number as before.

**Note:** The process will still operate properly with the correct numbers. The issue is that this is not being displayed to the user.

[Video of Tray Feed Count Issue]
(https://github.com/openpnp/openpnp/files/411648/Feed.Count.Bug.Video.zip)

vonnieda, 2016-08-10:
See ReferenceStripFeederConfigurationWizard for how to set up the bindings to make this work right.

vonnieda, 2016-08-11:
Fixed by #323

closed 2016-08-11 by vonnieda