openpnp/openpnp #328: Configure Camera Properties through machine XML

issus, 2016-08-19:
Camera properties should be configurable through the machine XML. A list of OpenCV property names and values should be able to be user configured, such as CAP_PROP_EXPOSURE for manually setting the exposure.

When the camera is initialised, these will then be iterated through, and sent to the camera with the .set(propId, value) method: http://docs.opencv.org/2.4/modules/highgui/doc/reading_and_writing_images_and_video.html#videocapture-set
A list of the property types is listed in the link above. Not all the parameters are supported by all cameras.

Further reading on manual exposure for the ELP cameras:
http://www.makehardware.com/2016/06/21/manual-exposure-vs-auto-exposure-for-elp-2-mp-usb-camera/
http://xn--1-2fa.fr/?p=110
http://answers.opencv.org/question/27182/fully-manual-web-cam/

As a note, .set(CAP_PROP_SETTINGS,1); should display the operating systems configuration window for the camera, which will show which features are supported by the cameras, as it appears to disable properties that are unsupported.

vonnieda, 2016-08-19:
Great idea!

vonnieda, 2016-12-30:
For reference, in OpenCV 2.4.13 the following capture drivers mention exposure:
> grep CV_CAP_PROP_EXPOSURE -rl *
cap_android.cpp
cap_dc1394_v2.cpp
cap_dshow.cpp
cap_giganetix.cpp
cap_intelperc.cpp
cap_libv4l.cpp
cap_pvapi.cpp
cap_v4l.cpp
cap_ximea.cpp

In OpenCV 3.2.0:
> grep CV_CAP_PROP_EXPOSURE -rl *
cap_aravis.cpp
cap_dc1394_v2.cpp
cap_dshow.cpp
cap_giganetix.cpp
cap_gphoto2.cpp
cap_intelperc.cpp
cap_libv4l.cpp
cap_pvapi.cpp
cap_v4l.cpp
cap_ximea.cpp

commit d8fba85d (2016-12-30): Adds support for setting opencv camera properties. Fixes #328.
commit a4e9b238 (2016-12-30): Merge branch 'develop' into feature/opencv-3.2.0-0

* develop:
  Trace open cv camera properties on initialize to make it easier to debug issues with this feature.
  Added a button to read the current property value on the open cv camera wizard.
  Adds support for setting opencv camera properties. Fixes #328.
  Testing setting camera width and height before and after open to see if it helps with #395.
  Added links to documentation in Help menu.
  Added support for boolean actuators in ReferenceAutoFeeder. Added support for post pick actuator in ReferenceAutoFeeder. Added post pick support to  [... cut]
commit 1a506d7a (2017-01-07): Merge branch 'develop' into feature/loose-part-feeder

* develop: (97 commits)
  Adds a new ScriptRun CvPipeline stage that allows for custom programming of CvPipelines.
  Rearranged baclash fields a bit to clean up the GcodeDriverWizard.
  made textfields a bit more narrow
  moved form elements into two grid rows
  documentation
  form and keyword fixes
  more blackslah compensation GUI process - still need to fix form
  Fixes #398: Feeder exceptions are no longer swallowed during job processing. When reporting there are no feeders available, if a feed was previously attempted we store the er [... cut]
closed 2016-12-30 by vonnieda