openpnp/openpnp #329: Fiducial recognition should use the Pipeline system, configured in Machine XML.

issus, 2016-08-19:
Currently the fiducial recognition does not use the OpenCV pipeline as other parts of the software does. 

The fiducial recognition should be ported over to use CvPipeline, with the default pipeline configuration matching the current class. 

Users will then be able to experiment with custom pipelines for improved fiducial matching where their machine requires it.

visualecho1, 2017-01-21:
If a button is added to the part alignment config wiz to use the default fiducial camera, that config can be used for the fiducial pipeline editing and storage (you might also have the Edit button switch cameras if the name is like FIDUCIAL-*).  How would I get an accessor to that pipeline data from within ReferenceFiducialLocator?  It looks like it's only accessible from ReferenceBottomVision, is there a way to link up to that instance?

Question from the other side: is there an example of template matching in a pipeline staged vision operation?  I'm getting better at finding circles, but this is a little harder than sprocket holes.

vonnieda, 2017-01-21:
@visualecho1 The pipelines would not be shared - they are too different. When we convert the fid system to use pipelines a pipeline will be added to the fid locator and it will be customized similar to how bottom vision.

There's currently no template matching examples with the pipeline system, though there is template matching available. See https://github.com/openpnp/openpnp/wiki/CvPipeline for information about how to experiment with pipelines outside of OpenPnP.

visualecho1, 2017-01-21:
Hi Everboddy! "When we convert the fid system" - is that on a schedule?  Is this where that's being discussed?  Would you mind if I tried to help with this?  The code is really close to working on my machine, so I've been making fixes and then I kinda noticed that a few of the things I've fixed were on the issues list.  I'm glad to hear that there's a team working on this.

Fiducials are parts (footprint to camera) which don't need to be aligned with bottom vision.  It definitely makes sense to keep the pipeline with the part, one per atomic footprint-camera.  A part is a fiducial when specially marked as a part (aka "FIDUCIAL-HOME") or only when it's attached to a board?  The model doesn't seem to work well with this.  I think that any part can be a fiducial, that being a fiducial (sh/c/w)ould be a boolean attribute of a part-package relationship.  As much as I think this world needs more "bottom vision", I think "part vision" might be a better term, select a required camera.

Okay, so I've got some definite ideas about the fiducial vision pipeline, could you tell me more about the conversion?  Where is that being discussed?  I'm somewhat eager to see this code working better in many respects, but I'm very new here, only been debugging this app for a week or so.

vonnieda, 2017-01-21:
@visualecho1 This is the right place to discuss the feature. There isn't a schedule, per se. You can see my backlog at https://waffle.io/openpnp/openpnp. This feature is currently fourth on the list. There isn't a team; it's just me :)

As you said, we currently consider something a fiducial when it is marked as such in a placement (type=Fiducial), or for the special FIDUCIAL-HOME part. For the record, I don't like the special FIDUCIAL-HOME name and intend to eventually just make GcodeDriver specify which part it wants to use for homing.

My plan was to just replace the hardcoded vision code in ReferenceFiducialLocator with a single CvPipeline. I don't really feel that it's important to have a unique fiducial pipeline per part, but I am open to changing my mind on that if there is value in it.

There isn't much more to say about the conversion, but it sounds like you have some bigger ideas on it, so let's discuss! This ticket is as good a place as any to discuss or if you'd like some greater exposure, post to the mailing list: http://groups.google.com/group/openpnp

Jason

visualecho1, 2017-01-21:
I had another coffee and decided I should soften my tone, thank you!

vonnieda, 2017-01-21:
Your tone is fine :) Sorry if I came off brusk. I would be happy to have
your help with this!

Jason

vonnieda, 2017-06-30:
I've pushed the first version of this to https://github.com/openpnp/openpnp/tree/feature/fid-pipeline and have also deployed it to the test branch, so it is available for download from http://openpnp.org/test-downloads/.

It works similar to bottom vision for configuration. You can configure the default pipeline in Machine Setup and then the pipeline on a part by part basis in the Parts tab. 

[image]

The default pipeline is identical to what we used to do internally. Any pipeline can be used as long as it returns a List<KeyPoint> in the results stage.

vonnieda, 2017-07-02:
This has now been merged to develop and released. Closing as finished.

closed 2017-07-02 by vonnieda