openpnp/openpnp #330: Strip feeder should use the CvPipeline for vision processing

issus, 2016-08-19:
The stripfeeder currently does not use the CvPipeline for recognition. As converting this completely would be a large task, I propose this is split into two stages.

**Stage 1**
Add pipeline for pre-processing the image, allowing it to be cleaned up for improved recognition by the current algorithm.

**Stage2**
Completely convert the vision system to use the CvPipeline.

Stage 1 would allow the user to configure steps to clean up the image in their Machine.xml file, but still use the current recognition code on the cleaned up image. Stage 2 would be a complete conversion to using the CvPipeline for recognition.

dzach, 2017-03-23:
What does the 'ready' label mean? Is this already implemented? Where can I find more info on how to tweak the strip feeder pipeline?

vonnieda, 2017-03-23:
Ready just means it's ready to be worked on. It's not done yet.

dzach, 2017-03-23:
Oh, ok! Is there any other way to tweak some parameters of the pipeline to suit different tapes/feeders? E.g. threshold per feeder?

vonnieda, 2017-03-23:
No, not currently. That's the purpose of this ticket - to add that functionality.

richard-sim, 2017-07-25:
If anyone interested in this feature could test out my pull request #604 and report back, it'd be appreciated.

richard-sim, 2017-07-29:
#604 has been updated with Stage 2 now completed also.

vonnieda, 2017-07-29:
Will check in the next few days, currently on vacation.

vonnieda, 2017-08-29:
Fixed in #604

closed 2017-08-29 by vonnieda