openpnp/openpnp #331: Improve handling of reaching the end of the strip feeder

issus, 2016-08-19:
When a StripFeeder reaches the end, and the vision system is unable to find another part the error below is displayed:
[image]

**Stage 1**
The wording on this is less than ideal. The error should be changed to "Feeder 0805: Unable to locate reference hole. The end of the strip may have been reached.", rather than asking the user a question.

**Stage 2**
Further to this, it should be further improved to allow the user to reset the feeder, however this is significantly more work.

The exception that is thrown would need to be a new class (such as FeederException,) inheriting Exception, which would reference the stripfeeder that failed. This would allow it to be handled differently to a generic exception. 

In handling the exception, a dialog popup along the lines of "Feeder 0805: Unable to locate reference hole. The end of the strip may have been reached. Would you like to reloaded it? [yes] [no]".
Perhaps a new dialog could then be displayed if the user clicks [yes], asking them to click OK once they have finished reloading the strip. This would then cause OpenPNP to re-initialise the strip feeder in question.

vonnieda, 2016-08-19:
This is still going to be a bit tricky. We may need to improve the Feeder
API to have a generic "reset()" function. Otherwise the UI is not going to
know how to call into the feeder to reset it.

I think a good first pass at your Stage 2 option is to instead just open
the feeder configuration panel for the user. There is existing code in
place that makes it easy to switch to the Feeders tab and select the right
feeder. See the Feeder Setup action on the Job Panel.

issus, 2016-08-19:
Perhaps my stage 2 can be stage 3, with your comment as the new stage 2?

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda