openpnp/openpnp #333: Make Logging Easier to Configure

vonnieda, 2016-08-27:
The logging system has a couple limitations that are starting to become serious problems:

* You must edit log4j.properties and restart OpenPnP to make changes to logging.
* The log4j configuration is complex and difficult to understand for non-programmers.
* The logging window #288 is at the mercy of the log4j configuration.
* Logging categories are too granular and make it hard to get all the information you need.

I think we need to make some improvements to logging in general:

* Replace log4j with something much simpler and which can be configured at runtime programmatically.
* Create a number of basic categories and change all existing loggers to use them. For example: driver, machine, vision, job, etc. These should be fairly general.
* Make the logging window #288 be the controller for the logging configuration. Add UI for setting category log levels and make it easy to override so that you can turn on full system tracing with one click.

In general, we want to make it very easy for users to turn on increased logging since this is usually required for submitting a bug report or help request. In fact, we may want to get rid of categories altogether and just use a single log level across the board as many UNIX programs do. e.g. -v, -vv.

This is related to #83 in that we need to be able to readily enable and collect logs.

cri-s, 2016-08-27:
When I was not able to configure it for the logging console on GUI,
I have asked you too, I simple switched to http://www.tinylog.org/ 
That worked without problem , finer configurability and some other advantage for the casual user like me.

vonnieda, 2016-08-27:
Thanks Cri, tinylog looks nice.

vonnieda, 2016-08-27:
tinylog does away with categories and per logger configuration entirely, which might be perfect for what we want to do. Instead of having configuration complexity, we could add some basic filtering to the logging window to make it possible for users to filter out things that annoy them.

vonnieda, 2016-10-28:
Mostly implemented in the feature/logging branch. Just need to remove the temporary wrapper I added.

commit f5d43b74 (2016-10-28): Fixes #333, logging is now handled by tinylog and uses a very simple default configuration. Log level can be managed in the log tab. Logs are captured to rotating files, similar to before, to the log window and to the console.
commit 1e1c8602 (2016-10-28): Merge branch 'feature/logging' into develop

* feature/logging:
  Fixes #333, logging is now handled by tinylog and uses a very simple default configuration. Log level can be managed in the log tab. Logs are captured to rotating files, similar to before, to the log window and to the console.
  Ripped out slf and log4j and replaced with tinylog + a small wrapper. Wrapper is just so I don't have to touch every class yet. Will be removed. Logging window now controls logging entirely.
commit 880019ef (2017-01-07): Merge branch 'develop' into feature/loose-part-feeder

* develop: (96 commits)
  Untitled machine setup tabs filled
  moving generic code for feeders to base class
  moved generic code for cameras to base class ReferenceCamera
  Feeders configuration tab in "Machine Setup" panel no titled with "Configuration"
  Configuration tabs within "Machine Setup" panel now with title
  Added further improvement to startup message by adding 'Namste' along with multi language 'Hello'
  Feeder panel layout in line with other panels
  executing "first pull" tutorial
  Improves error messages for fidicual det [... cut]
closed 2016-10-28 by vonnieda