openpnp/openpnp #335: two placement fiducial wizard ,Angle is wrong.

cri-s, 2016-08-29:
if board should  have angle 0 or 90, and wizard was run before with result of angle 0.666 and it is run again where
Board placed have angle 3.333 the resulting error of second fiducial is over 2.5mm on 200:200mm Fiducial.

Workaround: reset Angle to 0 / 90 degree before running wizard.
Severity: low
Fix: calculate idealA/B with 0,0,0,0 board location for Angle/offser

cri-s, 2016-08-29:
Additional side effects, if that was not evident.
Same error exist if running manual or automatic fiducials
and  then running assembly job that include fiducial check.

cri-s, 2016-08-29:
How to reproduce this error: (untested)
Load Board
Place existing board at 30 degree and 5 cm offset (x+y).
Run twoPosFiducialWizard
Check results. BoardLocation + move to Fiducial
Place existing board at 60 degree and 10 cm offset.
Run twoPosFiducialWizard
Check results.
Run twoPosFiducialWizard again and see results.

vonnieda, 2016-08-29:
Thanks @cri-s. I will try to reproduce this tonight and report back.

vonnieda, 2016-08-30:
Okay, I believe I have reproduced the issue, and I now understand it.

What it boils down to is that the input board location when calculating two point location or fid location should not affect the resulting location. With the two placements and the two actual placement locations, we have all we need to calculate the angle and offset. The input board location should just be used to determine the side, width and height of the board. If the input board location affects the results at all the function is not working correctly.

I have a test case written that seems to prove this, but it's late, so I will finish getting it figured out tomorrow and see if I can solve it.

@cri-s I understand now what you mean about using the centers of the fids. I think that that will be included in the actual solution.

vonnieda, 2016-09-05:
I now have a working fix for this problem in https://github.com/openpnp/openpnp/commit/f98144f680055500507f7a78bad08e94bcf12efc. The code is not yet integrated - too tired to do that right now. Hopefully will get it done tomorrow. The new code passes a test suite that includes 1M random locations, and also does not even take into consideration the incoming board location, which was really the problem all along.

vonnieda, 2016-09-05:
Note: this probably doesn't work with bottom side. Haven't tested it yet. Likely need to invert angle.

vonnieda, 2016-09-05:
This is now fixed, code is checked into develop. The two old functions are removed and a new one is in place that passes every test I can think of. @cri-s, please give this a try and see what you think.

closed 2016-09-05 by vonnieda

cri-s, 2016-09-05:
I think for bottom side, board size should be set using reasonable guess if
any board location X on bottom side is bigger as board width. Reasonable
guess could be max-X + 5 mm or +0.2" .  If not, and you explitly allow
negative coordinate, testcase should include that.
For offset, you should Use average from point a+b, otherwise if board is
bigger the offset error can be seen.
You could test it placing A4 or legal letter size paper as board, creating
2 fiducial, 0:0 and diagonal edge, use wiki to know exact paper size. If
you PNP don't allow that size draw fiducial point on paper.
Now rotate fictive board 20 degree, make fiducial check using wizard and
then go with camera to fiducial that is far from board zero and check
offset error.
This error could be avoided if instead of using offset distance of idealA
you use idealA.add(idealB).multiply(0.5,0.5,0.0,0.0)

vonnieda, 2016-09-05:
Cri, are you saying there is an error in the new code? Or are these
suggested improvements? If there is an error, can you supply input values
that show the error?

Thanks,
Jason

cri-s, 2016-09-06:
Code should work. This are improvements, but to 99.8% of users unnecessary,
sorry, delete this comments.

commit f64bde6c (2016-09-05): Fixes #335. This should resolve a whole host of issues people are having with fiducials and the two point location process. Primary fix is that ideal locations are now calculated with board location set to 0, and we use the actual calculated ideal board locations as our baseline.
Adds a new test that runs 100k random boardlocations through the new process.